ROBOTIS-GIT/DynamixelSDK #200: [3.5.4][Python][USB2AX][MX-64AT] Issue : Why the shift to pyserial instead of ctypes wrapper for python driver?

gauthamvasan, 2018-06-18:
I don't have any code issues or failures. I just had some questions and didn't know the best way to reach out to you folks!  I've been using an older version of the `DynamixelSDK` which uses ctypes wrapper on top of C for python. I was initially using [pypot](https://github.com/poppy-project/pypot) and then our own in-house pyserial driver. We found your C and ctypes based python driver to be way better and more reliable in serial communication. For example, this was a test I ran a couple months back - Within a for loop I wrote to the goal_position register and read the present_position register for a DXL MX-64AT. The baudrate was 1,000,000 and return delay time was set to zero. I ran the loop for 1 million iterations and measured the computation time required for each iteration of a write and read op within the loop. When comparing drivers, this is what I found:

[image]

The delays were caused by pyserial and sometimes the communication was unreliable. There were no delays when using C. But with pyserial, I've seen delays ranging from 10ms to 800ms. I have 2 question:

- Did you find pyserial to be comparable if not better than your C driver? If yes, what version of python and pyserial are you using?

Thanks a lot for your time! Appreciate the great work you are all doing!

robotpilot, 2018-06-20:
Hi @gauthamvasan,

WoW COOL! I am very interested in the results of your experiments. 
We appreciate your efforts and we will test this part and look for improvements.

Thank you so much! :+1:

robotpilot, 2018-07-11:
Hi @gauthamvasan,
I think I did not answer your question properly. I am sorry for the delay.
1. Did you find pyserial to be comparable if not better than your C driver?
- The reason we moved from C-based Python to native Python was not a speed issue, and we aimed to provide it purely in Python as requested by our customers.
2. If yes, what version of python and pyserial are you using?
- We used pyserial 3.0 at the time of development, which depends on the user's development environment.
I apologize again for the late reply and I will close this issue. You can reopen this issue at any time if you need it.

closed 2018-07-11 by robotpilot

gauthamvasan, 2018-07-12:
@robotpilot No worries. Thanks a lot for your response! It clears things up for me. My team and I are working on super repeatable robot learning experiments and it was possible only with your ctypes based python driver. It didn't work as well with our pyserial driver. I'll just continue using a particular version of your repo. The issue is closed from my end as well.