ROBOTIS-GIT/DynamixelSDK #201: the DXL ID is mixed up and cant be separated to run position

msaufyrohmad, 2018-06-20:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`
- I'm using `C` Language
- I'm using `OpenCR` 
- I'm using `XM540-W270-R`
- and I'm having issue. my motor cant run separately. i configure my 14 motors with 6 master and i only set the master goal position.

however, all master is running and moving in the same time. i dont know why. 

I am using usb_to_dxl in arduino and the c code is compile in pc and loaded into the motor.

the scan function in dxl_monitor also only detect a few motors out of 14.

i hope get some help

thanks

saufy

OpusK, 2018-06-21:
> i configure my 14 motors with 6 master and i only set the master goal position.

Does 6 masters mean OpenCR?
Are each of the six masters independent of the physical connection?

The specification of the serial device causes problems if multiple masters are physically connected.

If possible, let us know the connection map and it will be helpful for us.

msaufyrohmad, 2018-06-21:
Total i have 14 motors in series.  Id 1-14
With 6 master and the rest is slaves.  The configuration is done and
working. (tested using arduino sw).

Now i am using c api with protocol 2.0 sdk.  Dxl_monitor only detect a few
motors and when i set the position of motor 9 (the id that detected by dxl
monitor), all motors is moving.

So i dont know why not all motors is detected and why i cant only move
motor with id 9?

Thanks

Saufy

OpusK, 2018-06-21:
Is that what you said about [Driver Mode](http://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#drive-mode10)?

> So i dont know why not all motors is detected

Please ckeck that "dxl_monitor" can not search the motor even if you connect only the motor.

msaufyrohmad, 2018-06-21:
Strange behavior occur.  All id of master motor that connected in series
change to 9!

Is this the problem that u said before?

Help me

OpusK, 2018-06-22:
> Strange behavior occur.  All id of master motor that connected in series change to 9!

Are you saying that all the IDs of the six masters have been changed to 9, when you did nothing?

If you want help, please give me a detailed answer to the questions I asked you earlier.
The information most important to me at present is the physical connection information of each motor.
In my opinion, it would be helpful to simply draw a hardware connection diagram.

msaufyrohmad, 2018-06-22:
All the 14 motors is connected to one rs485 port on opencr.  In series.

The id is 1-14.
1 master
2 slave
3 master
4 slave
5 master
6 slave
7 slave
8 slave
9 master
10 slave
11 master
12 slave
13 master
14 slave

At first all motor is working.  Then i test using ros with package in
dynamixel wiki.  I notice there all motors is running simultinausly.

Then i recheck back the id.  All master node change to 9.

I dont know why

Thanks

Saufy

OpusK, 2018-06-22:
Can you give or share a link to the code you tested?

I am guessing that the ID has been changed at the same time as changing the ID by broadcast.

msaufyrohmad, 2018-06-28:
the id is change after i run this :

roslaunch my_dynamixel_workbench_tutorial single_manager.launch

and i notice in dmesg this message appear : segfault at 0 ip
00007f0480d3e04d sp 00007ffe210f1d30 error 4 in
libdynamixel_workbench_toolbox.so[7f0480d35000+12000]

thank  you

saufy

robotpilot, 2018-07-05:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-07-05 by robotpilot

msaufyrohmad, 2018-07-05:
I am trying a work around by writing own dxl package to talk directly with
dxl sdk.

Thnks

Saufy