ROBOTIS-GIT/DynamixelSDK #203: Trouble in finding library "dxl_x64_c" in Windows8 64 bit using Matlab R2016a

MiriamBoz, 2018-06-25:
Hi ROBOTIS-Leon,

I have problem in finding library "dxl_x64_c" in Matlab using both protocol 1.0 and 2.0.
The problem is this:

I have downloaded package "DynamixelSDK-3.3.3" to use it in Matlab to control my Dynamixel MX-64T.

I have followed the new e-manual link you gave to me to install SDK for Matlab using the following link:

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#function-reference 
[image]
[image]
[image]

When Trying to test the software with .m files (read_write.m), I received an error that "MinGW-w64C/C++ Compiler" is missing and another error (please find the attached image files).

I tried to install "MinGW-w64C/C++ Compiler" and since I use Windos8 64 bit and Matlab R2016a, I faced other problems. 

However after trying to resolve the issue, now I receive the other error message which indicates that libraries are not found. (image of the error is attached).

I have also tried to set the environmental variables using the following link 

http://support.robotis.com/en/techsupport_eng.htm#software/dynamixelsdk.htm

and there is no bin folder in package "DynamixelSDK-3.3.3" which is recommanded in e-manual, but there is bin folder with the related folder inside "dxl_sdk_win64_v2_02" package which is missing all the products for Matlab. Therefore I was not able to edit the system variable. 

Can you tell me why I receive the related error messages while testing the Matlab files? The main problem is that library "dxl_x64_c" can not be found. 

Thank you,

Miriam 

==================================================

ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

robotpilot, 2018-06-25:
Hi @MiriamBoz,
I apologize for any inconvenience it may have caused. I recently completed the LabVIEW test of Window 10 and the latest Dynamixel SDK. First of all, I would like you to download and test the latest Dynamixel SDK. It is 3.6.1, not Dynamixel SDK 3.3.3.

MiriamBoz, 2018-06-27:
Hi @robotpilot @ROBOTIS-Will 

I am really stuck with the problems I asked in my latest comments which are also copied here by Will.
Can anybody guide me what the real problem is!
There are contradictions in the Matlab codes. It can recognize PC by strcmp(computer, 'PCWIN64'), but can not assign the lib_name='dxl_x64_c'!
Also once I check libisloaded('dxl_x64_c') the answer is 1, the other time is 0!

@robotpilot  Can you really check the Matlab codes? I have tried with different Matlab versions. 
Also the latest version is Dynamixel SDK is 3.6.0 not 3.6.1 in Matlab. Again I emphasize that I work with Matlab not LabView. 

Thanks,
Miriam

robotpilot, 2018-06-27:
Hi @MiriamBoz,

I have tested again to make sure there are no real problems. My development environment is as follows:

- Windows 10
- Dynamixel SDK 3.6.0 / MATLAB
- MATLAB R2017b
- USB2Dynamixel2
- XM430-W210-T

I set up Dynamixel SDK and MATLAB in the above development environment by referring to the following e-Manual.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#matlab-windows

So, Finally, I ran the motor through the read_write.m example and it was fine like image below.

[image]

When you use MATLAB, please check again that you have set up the folder and library mentioned in the above e-Manual.

MiriamBoz, 2018-06-27:
Hi @robotpilot 

I discovered I had a little problem though :) ... I had to disconnect R+ Manager 2.0, so that the port was released and could be used by the code :D ... That's all!

Thanks for your excellent support and replies so far!

Regards,
Maryam

robotpilot, 2018-06-27:
Cool. If you are still confused, please feel free to let us know.
Good luck! :)

closed 2018-06-27 by robotpilot