ROBOTIS-GIT/DynamixelSDK #204: About movement speed control

flilledshark, 2018-06-26:
ISSUE TEMPLATE ver. 1.1.1

Hello.
I am planning to control DYNAMIXEL using SDK and USB 2 Dynamixel.

· DYNAMIXEL SDK - 3.6.0 is used.
· Currently DYNAMIXELAX - 12 is used.
· We are programing in C ++.

I write it in read_write which was in example and move AX.
There was no problem in moving the speed by a constant, but trying to change speed like a trigonometric function does not work.
I attach the code below.
I'm sorry, please let me know.

-------------------------------------------------
int main()
{
  // Initialize PortHandler instance
  // Set the port path
  // Get methods and members of PortHandlerLinux or PortHandlerWindows
  dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

  // Initialize PacketHandler instance
  // Set the protocol version
  // Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
  dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);

  int index = 0;
  int dxl_comm_result = COMM_TX_FAIL;             // Communication result
  int dxl_goal_position[2] = {DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE};         // Goal position
  double h;
  double a = 0, b = 0, c = 0.00;
  double x = 0;
  int dxl_goal_speed = 2*sin(x);
  clock_t start, end;

  uint8_t dxl_error = 0;                          // Dynamixel error
  uint16_t dxl_present_position = 0;              // Present position
  uint16_t dxl_present_speed;					//Present speed

  // Open port
  if (portHandler->openPort())
  {
    printf("Succeeded to open the port!\n");
  }
  else
  {
    printf("Failed to open the port!\n");
    printf("Press any key to terminate...\n");
    getch();
    return 0;
  }

  // Set port baudrate
  if (portHandler->setBaudRate(BAUDRATE))
  {
    printf("Succeeded to change the baudrate!\n");
  }
  else
  {
    printf("Failed to change the baudrate!\n");
    printf("Press any key to terminate...\n");
    getch();
    return 0;
  }

  // Enable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_ENABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    printf("Dynamixel has been successfully connected \n");
  }

  ofstream outputfile("test.txt");
  double d = 0, t;
  while(1)
  {
    printf("Press any key to continue! (or press ESC to quit!)\n");
    if (getch() == ESC_ASCII_VALUE)
      break;

	// Write goal speed

	dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, 1, ADDR_GOAL_SPEED, dxl_goal_speed, &dxl_error);
	if (dxl_comm_result != COMM_SUCCESS)
	{
		printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
	}
	else if (dxl_error != 0)
	{
		printf("%s\n", packetHandler->getRxPacketError(dxl_error));
	}

	Sleep(120);		
    // Write goal position
    dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_MX_GOAL_POSITION, dxl_goal_position[index], &dxl_error);
    if (dxl_comm_result != COMM_SUCCESS)
    {
      printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
    }
    else if (dxl_error != 0)
    {
      printf("%s\n", packetHandler->getRxPacketError(dxl_error));
    }
	Sleep(120);	

	start = clock();
    do
    {
      // Read present position
      dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, ADDR_MX_PRESENT_POSITION, &dxl_present_position, &dxl_error);
      if (dxl_comm_result != COMM_SUCCESS)
      {
        printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
      }
      else if (dxl_error != 0)
      {
        printf("%s\n", packetHandler->getRxPacketError(dxl_error));
      }

	  // Read Dynamixel#1 present speed
	  dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, 1, ADDR_PRESENT_SPEED, &dxl_present_speed, &dxl_error);
	  if (dxl_comm_result != COMM_SUCCESS)
	  {
		  printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));

	  }
	  else if (dxl_error != 0)
	  {
		  printf("%s\n", packetHandler->getRxPacketError(dxl_error));
	  }
	  end = clock();
	  t = end - start;
	  a = dxl_goal_position[index];
	  b = dxl_present_position;
	  c = dxl_present_speed;
	  x = x + 0.01;
	  
	  cout << t << " Goalpos:" << a << " Prepos:" << b << " Prespe:" << c << " Ispe:"<< x <<"\n";

	  outputfile << t << " Goalpos:" << a << " Prepos:" << b << " Prespe:" << c << "ideal spe:" <<  x << "\n";

    }while((abs(dxl_goal_position[index] - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD));

    // Change goal position
    if (index == 0)
    {
      index = 1;
    }
    else
    {
      index = 0;
    }
  }

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }

  //Close file 
  outputfile.close();

  // Close port
  portHandler->closePort();

  return 0;
}

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

robotpilot, 2018-06-26:
Hi @flilledshark,

Thank you for your consideration and supports. However, as we check your source code, it does not seem to be a problem with the Dynamixel SDK. We can provide support at the protocol level for our customers' use of Dynamixel, but we can't help you with your application software algorithm problems. I hope you will understand this.

flilledshark, 2018-06-26:
Thank you for answering.
I understand. SDK has no problem.
What I wanted to know was about algorithms.
I'm sorry.

closed 2018-06-26 by flilledshark