ROBOTIS-GIT/DynamixelSDK #205: [3.6.1][C++/Arduino][OpenCR][XM430-W210-R] Issue : Need a basic example to get started

eng-Sund, 2018-06-30:
I am a mechanical engineer with considerable programming knowledge but I am having a hard time figuring out how to get started with Dynamixel X series motors and Open CR 1.0. I am having difficulty understanding the differences between Dynamixel Workbench and SDK. As, in the Open CR manual, there is an basic example to talk to Dynamixel motors using the Workbench. The quad bot project that I am working on uses:

-DynamixelSDK ver. 3.6.1 
-C++/Arduino Language
-Open CR1.0 Board
-XM430-W350-R and XM540-W270-R motors
 
What functions do I use to access the data on the control table of the X series motors?
The examples provided with the SDK on github uses Visual Studio. Are there any examples for Arduino? Also, under the src folder, there are a bunch of .cpp files. Which file is the relevant one?
All I plan to do it read the position and set the final position (including setting velocity profiles etc.) for the various motors.

I apologize if I have not followed the format properly. This is my first ever port on github.
Thanks in advance for any help!

OpusK, 2018-07-02:
hi, @eng-Sund 

please refer to [E-manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/cpp/cpp_porthandler/#api-reference) for APIs

And, [this link](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/07.%20DynamixelSDK) for examples.

eng-Sund, 2018-07-03:
Hey, thanks a lot! I actually got the second link from the deleted read me file. It would be great for beginners like me if the read me file was present.

closed 2018-07-03 by eng-Sund