ROBOTIS-GIT/DynamixelSDK #206: [TxRxResult] Incorrect status packet!

serogamon, 2018-07-03:
ISSUE TEMPLATE ver. 1.1.1

Hello, i follow the steps from the tutorial video and my programm write this when i click on run:

Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Dynamixel has been successfully connected 
Press any key to continue! (or input e to quit!)

[ID:001] Present Position : 1806 	 [ID:002] Is Moving : 0
Press any key to continue! (or input e to quit!)

[ID:001] Present Position : 1809 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 1817 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 1824 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 1831 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 1838 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] groupBulkRead getdata failed>> 

When i erase the code for getdata then my servos works correctly but every few steps it writes "incorrect status packet" and the present position is "0" but after that the servos go on:

Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Dynamixel has been successfully connected 
Press any key to continue! (or input e to quit!)

[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 2291 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 0 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 0 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2271 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2267 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2262 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2259 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2254 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2250 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2245 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2241 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2236 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2231 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2227 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2222 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2217 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2212 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2207 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 0 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2189 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2182 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2177 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 0 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2166 	 [ID:002] Is Moving : 0
[TxRxResult] Incorrect status packet!
[ID:001] Present Position : 0 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2156 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2151 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2146 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2141 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2136 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2130 	 [ID:002] Is Moving : 0
[ID:001] Present Position : 2125 	 [ID:002] Is Moving : 0

I have absolutly no idea what this mean 

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

robotpilot, 2018-07-04:
Hi, @serogamon,
I am sorry for the inconvenience.
I need more details kinda environments like hardware settings, software source code (the original example filename you used). the number of motors in use. And similar problems can be caused by user's development environment and motor setting settings. There were similar issues in the following link. Have you checked?
https://github.com/ROBOTIS-GIT/DynamixelSDK/search?q=TxRxResult%5D+Incorrect+status+packet&type=Issues

serogamon, 2018-07-09:
Hi, i will reset the motor settings tomorrow. I keep you updated.

Thank you for the answer.

closed 2018-07-15 by robotpilot