ROBOTIS-GIT/DynamixelSDK #207: [3.5.4][MATLAB][USB2Dynamixel2][MX-64] Issue :

kinter107, 2018-07-07:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

robotpilot, 2018-07-09:
Hi @kinter107,

Thank you for sharing all the progress and its references.
When I tested it briefly, there was no problem getting the current position in velocity mode. 
Have you finished testing a single motor?

kinter107, 2018-07-10:
Hi @robotpilot,

Thanks for your feedback.
actually i'm working with 2 motors and my purpose is to control a 2 degree of freedom parallel robot Named TezGoz and you can Google it if you don't know it .
But for testing my code before implementing on the bot, i use one motor and there was no success sadly , I think the problem might be related to my code . 
Can you use My code And test it if you have access to any dynamixels ?!

robotpilot, 2018-07-11:
Hi @kinter107,
I modified it based on your code and tested it. With two motors in velocity mode.
I was able to get the value without any problem when I read the position value while rotating at the speed. You should refer to the code in the link below and try again.
https://gist.github.com/robotpilot/9bc2ba473ec8111c474ccbe209a500bb

kinter107, 2018-07-11:
Thanks alot @robotpilot
Tomorrow, i will go to university and test The code.
i will share the results with You here.

kinter107, 2018-07-14:
Hi again @robotpilot 
today i tested your code with 2 servos and it successfully worked, so checked it with mine and finally found out where was the problem !!
i should have put the following code between lines 115-116 :
`
% Add parameter storage for Dynamixel#1 present position value
dxl_addparam_result = groupSyncReadAddParam(groupread_p, DXL1_ID);
if dxl_addparam_result ~= true
  fprintf('[ID:%03d] groupSyncRead addparam failed', DXL1_ID);
  return;
end
`
Thanks alot man you helped me so far ...
i really appreciate for the time you spend to solve my problem.

robotpilot, 2018-07-15:
It is good to hear back from you!!!
Please, let me know if you have any questions on this issues.
Thanks. :)

closed 2018-07-15 by robotpilot