ROBOTIS-GIT/DynamixelSDK #209: [3.6.0][MATLAB][USB2Dynamixel2][MX-64] Groupsyncread function makes a huge delay.

kinter107, 2018-07-15:
Hi,
I am working on a robot which is called TezGoz ( you can google it to know how it works ) with following specification :
1 - consists of 5 revolute joints
2 - 2 Degree of Freedom ( rotation around X-axis and Y-axis )
3 - it can be used as a camera stabilizer
here is the picture of it :

[image]

To control this robot i use a gyroscope ( MPU 9051 ) located on End effector , and our goal is to keep End effector parallel to reference surface (earth surface ) no matter how many disturbances applied to the base.
the controller is simple PID (only P part ) and it controls the robot perfectly but the problem occurs when i want to read present Position and velocity of each servos.
without reading present position and velocity and just by sending control signals to servos the robot is controlled smoothly but when i add the Codes ( in matlab ) for reading current state of servos ( position and velocity ) there will be a huge delay as if the controller can't do anything.
at first i thought something is wrong with my code so i tried to rewrite my code this time only for one servo so the procedure is like this :
if x-axis reported be sensor is positive ----> speed will be +100
if x-axis reported be sensor is negative ----> speed will be -100
here is my code :

[onlymove_by_sensor.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2195634/onlymove_by_sensor.txt)

which uses another .m file with name "Opening"which is below and need to be the same place as the main code is :

[Opening.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2195636/Opening.txt)

if you comment lines 252 through 267 of first code everything works like a charm but when you add them the delay bothers.
any help will be greatly appreciated.

robotpilot, 2018-07-15:
Hi @kinter107,
I am so sorry, but I didn't understand your question. Do you need to modify your code? I would like to ask me more questions about what parts of the SDK are not working well.

kinter107, 2018-07-16:
Hi @robotpilot 
Thanks for feedback , let me explain it simple :
if you open the " Onlymove_by_sensor.txt " there is only one While loop inside it at which there are codes to first get data from Gyroscope , second Compute speed of the servo (which needs to be sent to servo ) and third read the current state of servo ( position and velocity ).
if i comment codes that reads the current state of servo , everything works just fine.
what i mean by fine is that as soon as X-axis reported by Gyro changes its sign , speed of servo will change from +100 to -100 or vice versa.
but if i uncomment those codes , the result is not as it was before and it seems that GroupSyncRead function makes delay .
for better understanding i uploaded 2 short videos :
part 1 ( everything is 👍 ) :

[part1-OK.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2196885/part1-OK.zip)

part 2 ( everything is 👎 ) :

[part2-NOT-OK.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2196886/part2-NOT-OK.zip)

robotpilot, 2018-07-17:
Hi @kinter107,
Since I use a different development environment with you, so I have commented on the Gyroscope related part and run it. I worked without problems. The delay you said did not occur. You can also comment on the Gyroscope source code. I think other problems may be the cause.

kinter107, 2018-07-17:
Thanks @robotpilot  for feedback, if the problem was due to gyroscope related part the code shouldn't run at all, but as you can see in those videos when i comment read current status related part everything works well.
and today i faced another problem while i was manipulating my code to see where might be wrong , and the problem was so weird !!
if you remember our previous discussion about reading present position in velocity mode, you modified my code so it worked well , but today i tested it again with 2 different speed , +20 and -20 , and something strange happened:
when i set goal velocity to +20 , code reads present position and velocity perfectly but when i change it to -20 , output is something like 4e9 for both present velocity and position!!!
any idea why this happens?

robotpilot, 2018-07-18:
It is not easy for us to modify the code of our users and help them. In particular, if a robot connected only to a motor can be reproduced, it can be tested. However, if use other external devices like other sensors, we can't reproduce the development environment.

kinter107, 2018-07-21:
Hi @robotpilot ,
i understand what you say , so i decided to remove Gyroscope related part.
in the following code i have just written a code which only reads present position and velocity of the servo, my problem is that when i set Goal velocity to +10 , everything is just fine, but when i change it to -10 present position is something like 4e9 and present velocity is 0!!
here is the code :

[Report.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2215965/Report.txt)

when goal velocity is +10 i get :

[image]

but when it is -10 i get :

[image]

i have reading problem when velocity is some negative numbers but when it is positive everything is ok

robotpilot, 2018-07-23:
Hi @kinter107
You can use the typecast used in MATLAB as shown below.
```
[typecast(uint32(dxl1_present_position), 'int32') typecast(uint32(dxl1_present_velocity), 'int32')]
```

kinter107, 2018-08-01:
Hi @robotpilot 
that solved my problem , and about the delay we discussed earlier i found out that it's because i'm using 2 ports at the same time in matlab and matlab is not powerfull enough to read from 2 ports in parallel.
i greatly appreciate your answer.

robotpilot, 2018-08-07:
Oh! GOOD!  I will close this issue. Please, let me know if you have any questions on this issues. :)

closed 2018-08-07 by robotpilot