ROBOTIS-GIT/DynamixelSDK #210: [3.4.1][C++][USB2Dynamixel2][AX-12A] Issue : The servos are too fast

KingKarlo30, 2018-07-16:
- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1]

- I'm using MatLab

- I'm using `USB2Dynamixel2` serial converter

- I'm using `AX-12A

  - The servos are too fast

Good Afternoon,

my name is Carlos Castellanos and im developing a control code for the phantom X using matlab, the code is already done and is working correctly, however the servos are moving too fast and the arm function is, in some point, dangerous.

We are trying to make the control table as following but we don´t know what other parameter we must introduce to minimize the speed.
`ADDR_MX_TORQUE_ENABLE       = 24;           % Control table address is different in Dynamixel model
ADDR_MX_GOAL_POSITION       = 30;
ADDR_DXL_GOAL_SPEED		    = 32;
ADDR_MX_PRESENT_POSITION    = 36;
ADDR_DXL_PRESENT_SPEED		= 38;

% Protocol version
PROTOCOL_VERSION            = 1.0;          % See which protocol version is used in the Dynamixel

% Default setting
DXL_ID1                      = 1;            % Dynamixel ID: 1
DXL_ID2                     = 2;
DXL_ID3                     = 3;
DXL_ID4                     = 4;
DXL_ID5                     = 5;
DXL_ID6                     = 6;
DXL_ID7                     = 7;
DXL_ID8                     = 8;
BAUDRATE                    = 1000000;
DEVICENAME                  = 'COM4';       % Check which port is being used on your controller
                                            % ex) Windows: 'COM1'   Linux: '/dev/ttyUSB0' Mac: '/dev/tty.usbserial-*'

                                            
                                            
TORQUE_ENABLE               = 1;            % Value for enabling the torque
TORQUE_DISABLE              = 0;            % Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 1023;          % Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 10;         % and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
DXL_MOVING_STATUS_THRESHOLD = 10;           % Dynamixel moving status threshold

ESC_CHARACTER               = 'e';          % Key for escaping loop

COMM_SUCCESS                = 0;            % Communication Success result value
COMM_TX_FAIL                = -1001;        % Communication Tx Failed

% Initialize PortHandler Structs
% Set the port path
% Get methods and members of PortHandlerLinux or PortHandlerWindows
port_num = portHandler(DEVICENAME);

% Initialize PacketHandler Structs
packetHandler();

index = 1;
dxl_comm_result = COMM_TX_FAIL;             % Communication result
% dxl_goal_position = [DXL_MINIMUM_POSITION_VALUE DXL_MAXIMUM_POSITION_VALUE];         % Goal position
dxl_goal_position_1 = 300;
dxl_goal_position_2 = 512;   % 
dxl_goal_position_3 = 512;
dxl_goal_position_4 = 512;
dxl_goal_position_5 = 512;
dxl_goal_position_6 = 512;
dxl_goal_position_7 = 512;
dxl_goal_position_8 = 800;
dxl_error = 0;                              % Dynamixel error
dxl_present_position = 0;                   % Present position

% Open port
if (openPort(port_num))
    fprintf('Succeeded to open the port!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to open the port!\n');
    input('Press any key to terminate...\n');
    return;
end

% Set port baudrate
if (setBaudRate(port_num, BAUDRATE))
    fprintf('Succeeded to change the baudrate!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to change the baudrate!\n');
    input('Press any key to terminate...\n');
    return;
end`

[image]

Please your help with this issue.

Thanks

robotpilot, 2018-07-17:
Hi KingKarlo30,
You can set the speed of the motor by changing the 'Moving Speed' value before moving to a specific position. The address of this control table is **32**. 
http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#moving-speed-32
The control table differs depending on the motor type, refer to the control table of AX12 below.
http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#control-table-of-ram-area

robotpilot, 2018-07-23:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-07-23 by robotpilot

lea30cr, 2021-12-13:
Hello everyone, I'm sorry i'm really bad at programming but I'm supposed to control an AX-12A in speed (reach a certain position by mooving at a certain speed), and this is what I write on arduino but it doesnt work:

  ax12SetRegister(id,32,100);
  SetPosition(id,1000);

the libraries : 
#include <ax12.h>
#include <BioloidController.h>

The motor goes to the right position but won't change the speed
I'm sorry if the question is stupid but i'm in mechanical engineering and i don't really know programmation

ROBOTIS-Will, 2021-12-15:
Hi @lea30cr 

No worries at all.
A couple tips for using GitHub issue:
1. Don't be afraid of creating a new issue thread unless you're experiencing the exact same issue of the thread.
2. Describing your environment is extremely helpful, such as; controller, power source, DYNAMIXEL model, software libraries.

The inquired AX12 library is not an official library distributed by ROBOTIS.
If you are new to the Arduino programming, I'd recommend using our official Arduino library below.
- [Library] DYNAMIXEL library for Arduino : https://github.com/ROBOTIS-GIT/dynamixelShield
- [Product] DYNAMIXEL Shield for Arduino Uno form factor : https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/
- [Product] DYNAMIXEL Shield for Arduino MKR form factor : https://emanual.robotis.com/docs/en/parts/interface/mkr_shield/

You may also find some useful video instructions from [ROBOTIS OpenSource channel](https://www.youtube.com/c/ROBOTISOpenSourceTeam/videos).
- [Video] DYNAMIXEL Quick Start Guide for DYNAMIXEL Shield for Arduino MKR Series : https://youtu.be/1Y7FLG1n-9k

**If you have any other questions, please use the issue thread of the DYNAMIXEL Shield repository below.**
https://github.com/ROBOTIS-GIT/DynamixelShield/issues
Thank you.