ROBOTIS-GIT/DynamixelSDK #219: About dynamixel sdk in ROS

YunchuZhang, 2018-08-07:
ISSUE TEMPLATE ver. 1.1.1

Hi dear, Recently, I am working on ROS with your product. I met a problem that  when I use catkin_make to make your ros package from here( https://github.com/ROBOTIS-GIT/DynamixelSDK),  it could build and source and run python document. But, when I try to  add c++ document from your c++ folder, it can not make and build.Also, I  added  : add_executable(executable_name ${SRCS1}) target_link_libraries(executable_name ${catkin_LIBRARIES}) in your  official ros package. But it shows the error that “”ld returned 1 exit  status” and can not find  dynamixel::PortHandler::getPortHandler(DEVICENAME) and  dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION); read_write.cpp:(.text+0x23f): undefined reference to dynamixel::PortHandler::getPortHandler(char const*)' read_write.cpp:(.text+0x250): undefined reference todynamixel::PacketHandler::getPacketHandler(float)' collect2: error: ld returned 1 exit status Can you help me figure it out? Thanks, Yunchu
--

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

YunchuZhang, 2018-08-07:
Thanks! But my question is that how can I utilize the dynamixel  package in ros when I create a new ros c++ package. At that package, I want to include your SDK as a library and write my own cpp to include it.As to the dynamixel_workbench, it writes in service and client format to control the motors. I want to control 5 motors directly in one cpp node. So, how can I write it? Or how can I utilize the attachment code in ros package?
[read_write.cpp.tar.gz](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2268382/read_write.cpp.tar.gz)

robotpilot, 2018-08-07:
You can install the `dynamixel_sdk` package as follows.
```
$ sudo apt-get install ros-kinetic-dynamixel-sdk
```
And then you can use the `dynamixel_sdk.h` file in your code as shown below.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/f96421ec1c97a7aa561f7ef563590c443128956b/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_driver.h#L30

The reason I introduced the `dynamixel_workbench` package and its manual is because I want to show you one example using dynamixel_sdk on ROS. The example of dynamixel_workbench uses service as you said, but you can control it internally without using it. If you refer to the materials I mentioned, I think you can make your own ROS package.

I can't modify your source code. I am sincerely sorry.

YunchuZhang, 2018-08-08:
I have seen the dynamixel_workbench package and manual, is there any examples that could control it directly like the c++/python example without input from command line? Thanks

robotpilot, 2018-08-08:
Among the example code we provide, the following code is the simplest example of using the `dynamixel_sdk` and `ROS`. 
There are two publishers, one subscriber, and one service. 
If you don't need these topics and service, you can delete and modify them.

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/position_control.cpp
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#position-controller

robotpilot, 2018-08-10:
I'm going to close this until you update the issue with more information. You can reopen this issue whenever. Thanks.

closed 2018-08-10 by robotpilot