ROBOTIS-GIT/DynamixelSDK #220: While setting up with MX106, the error message shown : [TxRxResult] There is no status packet!

ycean, 2018-08-08:
Hi,
Here are my test details:
Dynamixel model: MX 106
Connected with pc using USB2Dynamixel
Running with Ubuntu 16.04

The error message is:
**/catkin_ws/src/DynamixelSDK/c/example/protocol1.0/read_write/linux64./$ ./read_write
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)**

I am trying to setup my servo according to the manual : **http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/device_setup/**

I did change the define detail according to the control table for MX106 in my read_write.c
The coding below is my amendment for my read_write.c:

**#if defined(linux) || defined(APPLE)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif**

**#include <stdlib.h>
#include <stdio.h>
#include "dynamixel_sdk.h" // Uses Dynamixel SDK library**

**// Control table
#define ADDR_MX_TORQUE_ENABLE 34 // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION 30
#define ADDR_MX_PRESENT_POSITION 36**

**// Protocol version
#define PROTOCOL_VERSION 1.0 // See which protocol version is used in the Dynamixel**

**// Default setting**

**#define DXL_ID 1 // Dynamixel ID: 1
#define BAUDRATE 57600
#define DEVICENAME "/dev/ttyUSB0" // Check which port is being used on your controller**
// ex) Windows: "COM1" Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

**#define TORQUE_ENABLE 1 // Value for enabling the torque
#define TORQUE_DISABLE 0 // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE 0 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE 4095 // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD 10 // Dynamixel moving status threshold**

**#define ESC_ASCII_VALUE 0x1b**

Is there any solution for it?

Thanks.

robotpilot, 2018-08-08:
Hi @ycean,
Thank you for re-upload the issue. 
The TORQUE_ENABLE address of the `MX-106` is `24`. I think you misunderstood this part.
http://emanual.robotis.com/docs/en/dxl/mx/mx-106/#control-table-of-ram-area

robotpilot, 2018-08-08:
Most of these kinds of error phrases are related to your development environment and motor configuration. Most of them are simple problems, so please check below.
- Could you recheck if the Dynamixel settings (ID, Baudrate, etc)are the same as you set in the code?
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/device_setup/
- Did you put the power in the Dynamixel over 12V?

ycean, 2018-08-08:
Yes, I did, As i am using usb2dynamixel to connect MX106 to my pc. And i am running it with linux in ubuntu 16.04, which the installation driver is excepted.
#define DXL_ID **1** // Dynamixel ID: 1
#define BAUDRATE **57600**
#define DEVICENAME "**/dev/ttyUSB0**" 

I use the external power supply over 12V for the MX106 during the test.

is that the range of position value does the matter? I defined it in 0-4095, which the original value in the code is 40-4000.

#define TORQUE_ENABLE 1 // Value for enabling the torque
#define TORQUE_DISABLE 0 // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE **0** // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE **4095** // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD 10 // Dynamixel moving status threshold

#define ESC_ASCII_VALUE 0x1b

robotpilot, 2018-08-08:
Hmm ... honestly, I don't understand your situation. So far as you say, there should be no problem. Would you change your development environment? How about running the same read and write example with the Python example? Or how about doing it in windows?

ycean, 2018-08-09:
Hi,

I did all except do it in Windows, as I have the mission to do it with ROS ><""
I am wondering why does the error coming out as I follow exactly the same procedure in the manual, which I also think that shouldn't have problem. But the same error still coming out. Except doing it in Window, can you recommend what should I try?

ycean, 2018-08-09:
Thanks for your advise. I am trying on it ^^

ycean, 2018-08-09:
Hi, 
I got this error message while I try to launch the single_manager file.
[image]

routiful, 2018-08-09:
Give user permission to port by using below commands

```
$ sudo chmod a+rw /dev/ttyUSB0
```

ycean, 2018-08-09:
Hi, thanks for your quick reply. But i did ><""
[image]

routiful, 2018-08-09:
Would you try to change `/baud_rate` param to 1000000?

ycean, 2018-08-09:
yes, I did XD
But the error still come out ><""

routiful, 2018-08-09:
Do you know exact ID and Baudrate your Dynamixel??

ycean, 2018-08-09:
I just refer to the control table for my motor model, which is MX106 @@""
http://support.robotis.com/ko/product/dynamixel/mx_series/mx-106.htm

routiful, 2018-08-09:
OK, then can you copy and paste on terminal after type below commands?

```
$ dmesg | grep tty
```

ycean, 2018-08-09:
Yes,..
test@test-SF100:~/catkin_ws$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    1.001480] 00:01: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    4.133546] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0
test@test-SF100:~/catkin_ws$

routiful, 2018-08-09:
Sorry for inconvenience. 
Lastly, try to change `\scan_range` param to 100 and `\baud_rate` to 57600 and 1000000.

If it not working, please make firmware recovery your Dynamixel to use R+ Manager 2.0 (It is only working in Windows) I attached related link for you.

http://emanual.robotis.com/docs/en/software/rplus2/manager/#firmware-recovery

ycean, 2018-08-09:
Okay....It's still cant found..I will try to make the firmware recovery then. TT

Thanks again.

ycean, 2018-08-10:
Hi, after I done the firmware recovery, it is working fine!!!
At first I downloaded R+ Manager 2.0 from (http://www.robotis.us/roboplus2/) and installed, I couldn't find the model for MX-106 (2.0). Fortunately, it's appear after I update that software by clicking on the refresh button in the window of R+ Manager 2.0.
 
Thank you very much!

Yeoh.

robotpilot, 2018-08-12:
Hi, is this issue resolved?

ycean, 2018-08-13:
Hi, It is working after I did the firmware recovery. 
Thank you very much.

Yeoh.

robotpilot, 2018-08-13:
Hi @ycean 
It is good to hear back from you. I'm going to close this until you update the issue with more information. Thanks.

closed 2018-08-13 by robotpilot