ROBOTIS-GIT/DynamixelSDK #223: [3.6.2][C]/[Java]/[Python][USB2Dynamixel2][XH430-W350-R] DynamixelSDK still holds a reference to the port after it's closed (blocking it until reboot)

orgicus, 2018-08-14:
If I run the **readwrite SDK sample** which open the serial port then closes it that works.

The issue is I unplug/replug the USB cable **after** I've ran a DynamixelSDK sample once: the serial port disappears (from /dev/tty.*). 

What's weird is that the device is still present in System Information USB, even when the USB cable is unplugged.

It appears the FTDI driver somehow gets in a strange locked state as I can't unload/reload it:

```bash
sudo kextunload -b com.FTDI.driver.FTDIUSBSerialDriver
Password:
(kernel) Can't unload kext com.FTDI.driver.FTDIUSBSerialDriver; classes have instances:
(kernel)     Kext com.FTDI.driver.FTDIUSBSerialDriver class FTDIUSBSerialDriver has 1 instance.
Failed to unload com.FTDI.driver.FTDIUSBSerialDriver - (libkern/kext) kext is in use or retained (cannot unload).
```
If it helps here is some `udev` and `kexstat` info:
```bash
-o USB <-> Serial Converter@14100000  <class AppleUSBDevice, id 0x1000005e2, !registered, !matched, inactive, busy 2 (78721467 ms), retain 13>
      |       +-o USB <-> Serial Converter@0  <class AppleUSBInterface, id 0x1000005f2, !registered, !matched, inactive, busy 2 (78721016 ms), retain 10>
      |         +-o FTDIUSBSerialDriver  <class FTDIUSBSerialDriver, id 0x1000005f4, !registered, !matched, inactive, busy 1 (78720954 ms), retain 6>
      |             {
      |               "bInterfaceNumber" = 0
      |               "CFBundleIdentifier" = "com.FTDI.driver.FTDIUSBSerialDriver"
      |               "IOProviderClass" = "IOUSBInterface"
      |               "IOClass" = "FTDIUSBSerialDriver"
      |               "bConfigurationValue" = 1
      |               "IOProbeScore" = 100000
      |               "idProduct" = 24596
      |               "IOMatchCategory" = "IODefaultMatchCategory"
      |               "IOPowerManagement" = {"CapabilityFlags"=32768,"MaxPowerState"=1,"CurrentPowerState"=1}
      |               "bcdDevice" = 2304
      |               "idVendor" = 1027
      |             }
$ kextstat | grep FTDI
  149    0 0xffffff7f80d6d000 0x7000     0x7000     com.FTDI.driver.FTDIUSBSerialDriver (2.3) ECC3AF36-431D-370D-86F2-5237785E9CF8 <121 39 5 4 3 1>
$ kextstat | grep 121
  121    2 0xffffff7f80d5c000 0xe000     0xe000     com.apple.iokit.IOSerialFamily (11) 854221E2-E1EB-361A-98C6-DC6B7E1EB0F4 <7 6 5 4 3 1>
  122    0 0xffffff7f81b1a000 0xa000     0xa000     com.apple.iokit.IOBluetoothSerialManager (4.4.6f1) 57A03351-E0D2-348B-A442-C4F5B6C4310E <121 7 5 4 3 1>
  149    0 0xffffff7f80d6d000 0x7000     0x7000     com.FTDI.driver.FTDIUSBSerialDriver (2.3) ECC3AF36-431D-370D-86F2-5237785E9CF8 <121 39 5 4 3 1>
```

I've tested the SDK using:
 - the Java library and c library directly
- the C++ library
- the Python version (which uses pyserial)

The Java/C/C++ version displayed the same behaviours: once the port handler closed the port, unplugging/replugging the devices rendered it useless until reboot.

The Python version couldn't open the port until I manually changed the default baud rate from 1M to 57600.
Once that worked got the same issue after the port was closed,

What's interesting is that if I simply use pyserial directly to write bytes, not using DynamixelSDK at all, the serial port works as expected.
For example, if I send LED on, LED off to XH430-350-R with ID 3 as bytes:
```
[0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x01,0xEF,0x64]
[0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x00,0xEA,0xE4]
```

Python 3 example:
```python
#!/usr/bin/env python3
import serial
import time

ID_3_LED_ON  = [0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x01,0xEF,0x64]
ID_3_LED_OFF = [0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x00,0xEA,0xE4]

s = serial.Serial("/dev/tty.usbserial-FT2GZFM6",57600)

s.write(ID_3_LED_ON)
sleep(3)
s.write(ID_3_LED_OFF)

s.close()
```
I've tested with Python 2 as well:
```python
#!/usr/bin/env python3
import serial
import time

ID_3_LED_ON  = [0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x01,0xEF,0x64]
ID_3_LED_OFF = [0xFF,0xFF,0xFD,0x00,0x03,0x06,0x00,0x03,0x41,0x00,0x00,0xEA,0xE4]

s = serial.Serial("/dev/tty.usbserial-FT2GZFM6",57600)

for byte in ID_3_LED_ON:
    s.write(chr(byte))
sleep(3)
for byte in ID_3_LED_OFF:
    s.write(chr(byte))

s.close()
```

I can unplug/replug the U2D2 device and the serial port is still listed and works after running something the above. As soon as I run `DynamixelSDK/python/tests/protocol2_0/read_write.py`, hit ESC, then unplug/replug the U2D2 device it become unusable as a serial port until I reboot OSX.

As mentioned before, the same issue can be reproduced with the read/write sample in C / Java (using the same C library) and C++.

If the issue could be resolved with the C library that would be amazing (as I guess will also resolve the same issue in Java). Perhaps the extra messages the DynamixelSDK sends via pyserial could be used to identity the issue with the C library ?

kijongGil, 2018-08-17:
Hi @orgicus :)
I'm sorry for late reply. First, thank you for your contribution.
I tested your idea and confirmed that it works as you say.
I think the attempt to solve this problem with pyserial is good. However, because this package provides libraries for each language, it seems that introducing pyserial does not fit the concept.

Thanks,
Gilbert

orgicus, 2018-08-18:
Hi @kijongGil ,

No worries about the delay :)

I have another update. I've been in touch with the FTDI support team and spotted this video as well. Both sources pointed to a solution that did work on my machine running OSX 10.11.6.
The solution was to **completely** remove the FTDI drivers and use the AppleUSBFTDI driver.
In case this may help other uses, here are a few links:

- [FTDI Drivers and Mac video from Alorium Tech](https://www.youtube.com/watch?v=e9sc72x59cQ)
- [Application Note AN_134 FTDI Drivers Installation guide for MAC OS X(pdf)](http://www.ftdichip.com/Support/Documents/AppNotes/AN_134_FTDI_Drivers_Installation_Guide_for_MAC_OSX.pdf) - handy especially for section "4 Uninstalling FTDI Drivers on OS X"
- [How to enable the root user on your Mac or change your root password](https://support.apple.com/en-us/HT204012)

These are steps I took for my setup, there might be slight differences on other machines:

1. Open **Directory Utility** (use Spotlight or open from `/System/Library/CoreServices/Applications/Directory Utility.app`)
2. Unlock (using system password)
3. Go to **Edit > Enable Root User** and setup your root password

At this stage you should be able to use `su`:
1. Login as root user: `su` (and use newly setup root password)
2. Remove FTDIUSBSerialDriver.kext from /System folder:`rm -r /System/Library/FTDIUSBSerialDriver.kext`
3. Remove FTDI drivers from /Library/Extensions:
```bash
rm -r /Library/Extensions/D2xxHelper.kext
rm -r /Library/Extensions/FTDIKext.kext
rm -r /Library/Extensions/FTDIUSBSerialDriver.kext
```
(Note that you might only find `FTDIUSBSerialDriver.kext` and not `D2xxHelper.kext` or `FTDIKext.kext` if you weren't already using both the FTDI Chip VCP and D2XX drivers)
Reboot if the U2D2 was already locked.

HTH,
George

kijongGil, 2018-08-22:
Thank you very much for your contribution. This is obviously useful for people with these problems.
I'm going to close this issue. Anyone can reopen if they have a problem.

Thanks,
Gilbert

closed 2018-08-22 by kijongGil