ROBOTIS-GIT/DynamixelSDK #225: [3.6.0][Arduino][OpenCR][XM430-W350-R] Issue : Questions about Sync Write

eng-Sund, 2018-08-17:
I have 2 questions about sync write:

1. If I sync write multiple motors with different angles and same speed (they stop moving at different times) then before the last moving motor stops, the program moves onto the next stage. 

For example, if I have two motors in a kinematic linkage: Motor 1 drives link 1 and Motor 2 is attached at the end of link 1 and drives link 2. Lets say in the first part of the code, motor 1 moves 180 degrees clockwise and motor 2 moves 45 degrees counterclockwise (at the same speed). And in the second part, motor 1 moves 180 degrees counter clockwise and motor 2 moves 45 degrees counterclockwise. So, before motor 1 completes 180 clockwise movement, it starts moving in the opposite direction as the program moves onto phase 2. Hence, I need to add a delay to allow for the motion to be completed. Is this normal or am I doing something wrong? It just seems like a hassle to add so many delays. I don't want the code to progress before all of the motion is completed.

Sample code:

// Allocate goal position value into byte array
      param_goal_position[0] = DXL_LOBYTE(DXL_LOWORD(stand2walk[1][0]));
      param_goal_position[1] = DXL_HIBYTE(DXL_LOWORD(stand2walk[1][0]));
      param_goal_position[2] = DXL_LOBYTE(DXL_HIWORD(stand2walk[1][0]));
      param_goal_position[3] = DXL_HIBYTE(DXL_HIWORD(stand2walk[1][0]));

      dxl_addparam_result = groupSyncWrite.addParam(lfh_id, param_goal_position);
      if (dxl_addparam_result != true)
      {
        Serial.print("[ID:"); Serial.print(lfh_id); Serial.println("groupSyncWrite addparam failed");
        return;
      }

      param_goal_position[0] = DXL_LOBYTE(DXL_LOWORD(stand2walk[1][1]));
      param_goal_position[1] = DXL_HIBYTE(DXL_LOWORD(stand2walk[1][1]));
      param_goal_position[2] = DXL_LOBYTE(DXL_HIWORD(stand2walk[1][1]));
      param_goal_position[3] = DXL_HIBYTE(DXL_HIWORD(stand2walk[1][1]));

      dxl_addparam_result = groupSyncWrite.addParam(lfk_id, param_goal_position);
      if (dxl_addparam_result != true)
      {
        Serial.print("[ID:"); Serial.print(lfk_id); Serial.println("groupSyncWrite addparam failed");
        return;
      }

      param_goal_position[0] = DXL_LOBYTE(DXL_LOWORD(stand2walk[1][2]));
      param_goal_position[1] = DXL_HIBYTE(DXL_LOWORD(stand2walk[1][2]));
      param_goal_position[2] = DXL_LOBYTE(DXL_HIWORD(stand2walk[1][2]));
      param_goal_position[3] = DXL_HIBYTE(DXL_HIWORD(stand2walk[1][2]));

      dxl_addparam_result = groupSyncWrite.addParam(lfa_id, param_goal_position);
      if (dxl_addparam_result != true)
      {
        Serial.print("[ID:"); Serial.print(lfa_id); Serial.println("groupSyncWrite addparam failed");
        return;
      }

      // Syncwrite goal position
      dxl_comm_result = groupSyncWrite.txPacket();
      if (dxl_comm_result != COMM_SUCCESS) Serial.print(packetHandler->getTxRxResult(dxl_comm_result));

      // Clear syncwrite parameter storage
      groupSyncWrite.clearParam();
      delay(2000);

2. How do I pass the GroupSyncWrite instance as a function argument. I want to shorten my code and I wanted to create functions to perform repeated sections of the code. For example, I used the following function to set the value of velocity and acceleration. So, this way I don't have to write the code over and over again. All I do is call the function.

void setProfVelAcc(dynamixel::PacketHandler *packetHandler,dynamixel::PortHandler *portHandler, uint8_t id, uint16_t address, uint32_t data, char indentifier)
{
  String s="";
  dxl_comm_result = packetHandler->write4ByteTxRx(portHandler, id, address, data, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.print(packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.print(packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    if(indentifier=='a')
    {
      s="Acceleration";
    }
    else
    {
      s="Velocity";
    }
    Serial.print("ID:"); Serial.print(lfk_id); Serial.print(" Dynamixel Profile "); Serial.print(s); Serial.println(" has been successfully set");
  }
}

I want to do something similar with the GroupSyncWrite instance, 'groupSyncWrite' but do not know how to pass it as a function argument. This way I will be able to pass the position values into a function which writes them into multiple motors.

dynamixel::GroupSyncWrite groupSyncWrite(portHandler, packetHandler, goal_pos_ad, goal_pos_len);

I hope my questions are clear. Thanks!

kijongGil, 2018-08-22:
Hi @eng-Sund :)
I'm sorry for late reply.
Refer to https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/171abd0bc92ebfe77c6a1c0e75b0c4215200ce45/c%2B%2B/example/protocol2.0/sync_read_write/sync_read_write.cpp#L292
If you change 'and' to 'or' here, it will be solved.

eng-Sund, 2018-08-24:
@kijongGil Thanks for the reply! How about part 2 the question? I cant even write this line of code globally: 
`dynamixel::GroupSyncWrite groupSyncWrite(portHandler, packetHandler, goal_pos_ad, goal_pos_len);`

How do I pass the groupSyncWrite instance as a function argument?

eng-Sund, 2018-08-30:
Hi,

Please let me know if you are unable to understand my question.

kijongGil, 2018-08-31:
Hi @eng-Sung.
I'm sorry for late reply and don't understand you question.
Could you tell me more?

robotpilot, 2018-09-06:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue.

closed 2018-09-07 by robotpilot