ROBOTIS-GIT/DynamixelSDK #226: [3.5.2][MATLAB][USB2Dynamixel2][XL-320,AX-12A] Issue: Noisy read2ByteTxRx when reading from multiple motors

dmrobotix, 2018-08-20:
When using read2ByteTxRx to read values like present position, goal, torque, or voltage, the first motor is always very noisy and gives mainly bad data. The motors after that give good data. I don't understand why this is happening or how to correct it. This happens using either USB2Dynamixel or USB2Dynamixel2 and also with AX-12A and XL-320 motors.

For example, I read the values like this in my MATLAB code:
` groupPos(1,i) = read2ByteTxRx(port_num, PROTOCOL_VERSION, LATERAL4, ADDR_PRESENT_POSITION);
    groupPos(2,i) = read2ByteTxRx(port_num, PROTOCOL_VERSION, LATERAL5, ADDR_PRESENT_POSITION);
    groupPos(3,i) = read2ByteTxRx(port_num, PROTOCOL_VERSION, LATERAL6, ADDR_PRESENT_POSITION);`

The values saved from motor, LATERAL4, will all be bad, but the rest of the motors will have reasonable values. Any idea what might be wrong? Thank you.

kijongGil, 2018-08-22:
Hi @dmrobotix :)
I tested to read present position value using read2ByteTxRx. I inserted the code below into this line.
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/171abd0bc92ebfe77c6a1c0e75b0c4215200ce45/matlab/protocol2.0/sync_read_write.m#L230
`
data_1 = read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL1_ID, ADDR_PRO_PRESENT_POSITION);
data_2 = read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL2_ID, ADDR_PRO_PRESENT_POSITION);
`
Both have good data.

Could you tell me about more a your situation?

Thanks,
Gilbert.

robotpilot, 2018-08-28:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

dmrobotix, 2018-10-01:
Hi, sorry for the late reply. I didn't get a notification that someone had responded. I haven't been able to test it recently, but I have tried it with two different robots. One running AX-12A motors and another running XL-320 motors. I read from the memory during a while loop. Were you able to do it in a while loop without error?

Thanks.

kijongGil, 2018-10-05:
I'm sorry for late reply.
Could you tell me your test environment and source code? I will reproduce your situation.

kijongGil, 2018-10-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-10-16 by kijongGil

dmrobotix, 2018-10-18:
No problem. Thanks, I'll get back to you soon.