ROBOTIS-GIT/DynamixelSDK #227: [3.6.0][Arduino][OpenCR][XM430-W350-R] Issue : Custom (Sinusoidal) Velocity Profile Implementation

eng-Sund, 2018-08-20:
Hi,

How do I implement a custom velocity profile? Especially a sinusoidal one. I have a couple of connected linkages whose joints are controlled by dynamixels. I want to vary the speed of the end-effector dynamixel based on the speeds of previous motors. I basically have a kinematic equation which gives me the relation between different angular velocities, but I don't know how to implement it.

Simplified Example: There are two dynamixels, d1 and d2. Velocity of d1=v1 and its present position = Theta 1 and Velocity of d2=5 x v1 x cos (Theta 1). How do I implement v2 for d2?

My implementation could be thought of similar to if you want to make a humanoid robot stand up. Assuming the hip moves straight up, and the feet are not moving, you can only rotate the joint motors at a certain relative speeds to satisfy the geometric conditions.

I hope the question in clear. Thanks!

Best,
Prasanna

kijongGil, 2018-08-21:
Hi @eng-Sund :)
This will be help you.
http://emanual.robotis.com/docs/en/platform/manipulator_h/manipulator_sdk_programming/#velocity-profile

Thanks,
Gilbert.

eng-Sund, 2018-08-21:
Hi @kijongGil,

Thanks a lot for the reply! I did not quite understand how I can generate sinusoidal trajectory. The link mainly talks about trapezoidal. Also how do I access the TrajectoryGenerator class for Arduino? Is there a class I can use for Inverse Kinematics?

kijongGil, 2018-08-24:
Hi again.
Unfortunately, TrajectoryGenerator is not accessible and there is no class for IK.
But, we will release IK using OpenCR with open_manipulator. I think open_manipulator is what you want.
Thnaks,

eng-Sund, 2018-08-24:
Thanks for the reply!
What is open_manipulator? How do I use it?

Best,
Prasanna

gpwhd07, 2018-08-27:
Hi @eng-Sund, 
You can refer to [here](http://emanual.robotis.com/docs/en/platform/openmanipulator) for open_manipulator.
http://emanual.robotis.com/docs/en/platform/openmanipulator.
Here, we mainly explain open_manipulator using ROS, but it is also including to use OpenCR.

Unfortunately, we are now updating source codes for open_manipulator used OpenCR.
We will release that to next week.
If you want to access now, you can access from "feature-open-manipulator" branch in ROBOTIS-GIT/OpenCR git. 

This will be help you.
https://github.com/ROBOTIS-GIT/OpenCR/blob/fb8a9de3ea762a6e5cc4500c842f86dab32316ff/arduino/opencr_arduino/opencr/libraries/OpenManipulator/example/Arduino/Test/kinematics/kinematics.ino
This is an example of using our kinematics library in the "feature-open-manipulator" branch.

closed 2018-08-30 by eng-Sund