ROBOTIS-GIT/DynamixelSDK #228: [3.6.2][C++][USB2Dynamixel][XM430-W210-T] Issue : [TxRxResult] There is no status packet!

AminMohammadi, 2018-08-21:
Hi
I'm using C++ in Ubuntu 16.04 with 6.4 g++ compiler, and I have the following problem. I wrote a code that can be compiled and run for 3 seconds, but suddenly I get this error that there is no status packet, and next line it says groupSyncReadPosition getdata failed. Here I upload the code and error message. As you can see it was able to run the code for 2.9 sec and print the data, but later on it says there is no status packet. I can run same code with same compiler version in HP laptop, but I cannot run it on ASUS. What's the problem?

[image]

[code.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2308020/code.txt)

kijongGil, 2018-08-22:
Hi @AminMohammadi :)
The code is the same, but if an error occurs depending on the PC, I suspect a baud rate problem.
How about change baud rate to 57600?

AminMohammadi, 2018-08-23:
Hi @kijongGil 
Thanks for your response. I changed the baud rate to 57600, but it didn't solve the issue. The same error happened.

kijongGil, 2018-08-24:
Did you use same Dynamixel? I mean, if you connected each Dynamixel to each laptop, Try to update Dynamixel firmware. 
Thanks.

AminMohammadi, 2018-08-24:
I updated Dynamixel firmware, and I got new error. Also the frequency of the motor were 330 Hz before using low latency, but it's now slower using low latency.

[image]

kijongGil, 2018-08-31:
I'm sorry for late reply. This error occurs when the communication is timeout.
How about change baud rate or delay?

closed 2018-09-07 by robotpilot