ROBOTIS-GIT/DynamixelSDK #229: [3.5.4][Arduino][OpenCM904] / [OpenCM 485] [MX-64] 2.0  Issue : Velocity Control Mode Code

mmartze, 2018-08-23:
I'm using :
- `DynamixelSDK ver. 3.5.4`

- [Arduino]

-  [OpenCM904] / [OpenCM 485] 

- [MX-64] 

- PROTOCOL 2.0

I'm having an `issue` like 
- I am trying to adjust the protocol 2.0 sample code to run the MX-64 on a wheel mode - Velocity control code. 
I managed to change the operating mode to 3 but the motor does not move. 
Is there a simple sample code for this?

Here my current code:

#include <DynamixelSDK.h>

#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

#define ADDR_PRO_GOAL_VELOCITY          104
#define ADDR_PRO_OPERATING_MODE          11

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        57600
#define DEVICENAME                      "3"                 // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0"

#define TORQUE_ENABLE                     1                   // Value for enabling the torque
#define TORQUE_DISABLE                    0                   // Value for disabling the torque

#define DXL_OPERATING_MODE                 1
#define DXL_GOAL_VELOCITY                 900

#define ESC_ASCII_VALUE                 0x1b

dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler;

int dxl_comm_result = COMM_TX_FAIL;
uint8_t dxl_error = 0;
uint8_t dxl_mode = 1;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while(!Serial);

  Serial.println("Start..");

  uint8_t dxl_mode = 1;

  
  // Initialize PortHandler instance
  // Set the port path
  // Get methods and members of PortHandlerLinux or PortHandlerWindows
  dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

  // Initialize PacketHandler instance
  // Set the protocol version
  // Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
  dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);

// Open port
  if (portHandler->openPort())
  {
    Serial.print("Succeeded to open the port!\n");
  }
  else
  {
    Serial.print("Failed to open the port!\n");
    Serial.print("Press any key to terminate...\n");
    return;
  }

  // Set port baudrate
  if (portHandler->setBaudRate(BAUDRATE))
  {
    Serial.print("Succeeded to change the baudrate!\n");
  }
  else
  {
    Serial.print("Failed to change the baudrate!\n");
    Serial.print("Press any key to terminate...\n");
    return;
  }

    // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_PRO_TORQUE_ENABLE , TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    packetHandler->getTxRxResult(dxl_comm_result);
  }
  else if (dxl_error != 0)
  {
    packetHandler->getRxPacketError(dxl_error);
  }
  else
  {
    Serial.print("Dynamixel is in Wheel Mode \n");
  }

      // OPERATING MODE
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_PRO_OPERATING_MODE , DXL_OPERATING_MODE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    packetHandler->getTxRxResult(dxl_comm_result);
  }
  else if (dxl_error != 0)
  {
    packetHandler->getRxPacketError(dxl_error);
  }
  else
  {
    Serial.print("Dynamixel is in Wheel Mode \n");
  }

}

void loop() {
      // Write goal Velocity
    dxl_comm_result = packetHandler->write4ByteTxRx(portHandler, DXL_ID, ADDR_PRO_GOAL_VELOCITY, DXL_GOAL_VELOCITY, &dxl_error);
    if (dxl_comm_result != COMM_SUCCESS)
    {
      packetHandler->getTxRxResult(dxl_comm_result);
    }
    else if (dxl_error != 0)
    {
      packetHandler->getRxPacketError(dxl_error);
    }
    else
    {
      Serial.print("moving\n");
    }
}

KurtE, 2018-08-24:
@mmartze and @kijongGil 

If I am reading this correct (from the Forum posting: http://en.robotis.com/service/forum_view.php?slg=&page_type=&bbs_no=2406936)

I believe you are trying to set the velocity to 900 and the default limit I believe is 435 from looking at the EEPROM register table: http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/#velocity-limit

KurtE, 2018-08-24:
@mmartze 

Looks like this program suffers from the same issue as another one that was up in a [forum thread](http://en.robotis.com/service/forum_view.php?bbs_no=2406928&page=2&save_sca=&sca=&save_stx=&stx=&sfl=)
that I debugged a few days ago... Should have checked, but @ROBOTIS-Will  did note it yesterday on the forum.  

The first main issue is, that you have two global variables: 
```
dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler;
```

Which never get set so when you reference them in loop(), the program will fault.  

This is because in setup(), you defined two local variables by the same names: 
```
dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

// Initialize PacketHandler instance
// Set the protocol version
// Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);
```
So all of the call offs you do in setup will use these local variables...

To fix this, change these two variable definitions into simple assignments of the global variables.
```
  portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

// Initialize PacketHandler instance
// Set the protocol version
// Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);
```
Again there may be other issues, like the first thing I noticed, but. hope that helps. 
And again it was found by @ROBOTIS-Will

ROBOTIS-Will, 2018-08-27:
Hi all,
Please refer to the forum thread below for the full context as Kurt mentioned in the above reply.
There were lots of discussions in the thread :)

[http://en.robotis.com/service/forum_view.php?bbs_no=2406936&page=1&save_sca=&sca=&save_stx=&stx=&sfl=](http://en.robotis.com/service/forum_view.php?bbs_no=2406936&page=1&save_sca=&sca=&save_stx=&stx=&sfl=)

closed 2018-09-03 by robotpilot