ROBOTIS-GIT/DynamixelSDK #230: DynamixelSDK on cRio using NI 9871

tpowell1591, 2018-08-27:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

I am using DynamixelSDK ver. 3.5.4 in Labview with a cRIO and an NI 9871. I am having the issue of the SDK library for the dynamixelSDK will not load properly on the cRIO. I am looking on advice on whether there should be an issue with the repository for this application. The NI 9871 is an RS485 board in the cRio but currently I cant even deploy my project onto the cRio. 

I have not modified any of the library files currently so they are the same as what you download in the SDK as of 8/27/2018.

Thanks for any advice and help.

Thomas

robotpilot, 2018-08-27:
Hi @tpowell1591,

We hope you understand that it is difficult to support specific hardware. My recommendation is to run the example on a regular PC first and check it. Next, we recommend that you try to run on specific hardware.

tpowell1591, 2018-08-28:
Hi @robotpilot     Nice name!

So the library runs fine on my computer, I just was wondering if there was a reason why it wouldnt run on the cRio. I understand supporting specific hardware is difficult and not worth your time whatsoever. But as far as i understand there is no labview difference between a PC and cRio.  Also I have run the library on my computer and it runs without error, it just pushes an error on cRio. 

I will keep charging forward and see what I can do, thanks for your advice and time!

Thomas

tpowell1591, 2018-08-28:
So after looking at another dynamixel library, i needed to import the library using a .dll and a .h file. Do you guys have a .h file for your labview DynamixelSDK? If you do that may be the only thing missing for it to work on cRio. Hope to hear from you soon!

Thomas

robotpilot, 2018-08-28:
Hi @tpowell1591,
I am truly sorry to you. I like the LabVIEW programs and I've used CompactRIO a long time ago. If I had a cRIO I would have tested immediately to find out why this issue occurred and to help you. But I don't have cRIO. But I will do my best to support you if there is anything I can do.
The header file is included in the downloaded Dynamixel SDK as described in the manual link below. Perhaps you have seen the setup associated with the path during installation.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/labview_windows/#labview-windows
The direct path is as follows:
https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/master/c/include/dynamixel_sdk

tpowell1591, 2018-08-30:
So after alot of debugging and head scratching I am going to be starting from scratch for my cRio version to work. So from that, can you give me an idea on how to calculate how many parameters there will be in any given message. I understand that every type will have anywhere from 0 to quiet a few depending on what your writing. But it doesnt give you a way to calculate how many parameters that each type should have or an easy way to calculate it. 

Thanks for your help.

robotpilot, 2018-09-03:
How are you doing? I hope everything is going great.

tpowell1591, 2018-09-04:
It is going I have created the packet creation package but I am stuck with your CRC calulator. It says at the top it is a CRC-16 (IBM/ANSI) but we cant figure out using the crc calculator in that you show how it works and how it is calculated. We tried calculating with the ICM standard but that was not working either.

robotpilot, 2018-09-18:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2018-09-18 by robotpilot