ROBOTIS-GIT/DynamixelSDK #232: New XL430-W250-T DOES NOT MOVE

Larcay, 2018-09-05:
Just bought a XL430-W250-T actuator and i can't make it move. It responds to REBOOT and FACTORY RESET instructions but it doesn't move at all.

Anyone have a clue of what can be happening?

I'm just passing the packets data using a general usb to serial converter.

kijongGil, 2018-09-06:
Hi @Larcay :)
Could you check your DXL is working by R+ manager?
http://emanual.robotis.com/docs/en/software/rplus2/manager/

Larcay, 2018-09-06:
I tried but could not connect once half duplex was not yet implemented. By now i could only test it transmitting packets via a simple Serial Port Terminal using CP2102. As i said, it worked on REBOOT and FACTORY RESET commands but dynamixel did not reacted by setting GOAL POSITION. 

CRC values wore validated and dynamixel was operating under 12V.

kijongGil, 2018-09-07:
I think you need to implement half duplex. Refer to https://www.instructables.com/id/How-to-drive-Dynamixel-AX-12A-servos-with-a-Raspbe/
And, if you do not have a reason to use cp2102, I recommend using U2D2(http://www.robotis-shop-en.com/?act=shop_en.goods_view&GS=3288&keyword=u2d2).

Thanks,
Gilbert

Larcay, 2018-09-14:
Feedback: I was not setting torque enable register and that was the issue. Dynamixel moved correctly after that