ROBOTIS-GIT/DynamixelSDK #233: DynamixelSDK feature request for readTxRx and writeTxRx

mtrossen, 2018-09-07:
Hello,

I was asked to submit this feature request here.

We are currently writing our firmware for the CM9.04 controller to run our robotic arms and we use protocol 2. In order to read and write data we have to send the register byte length over in the serial packets in order to do an 'if then' statement to decide which function to use.

`

            if intByteCount == 1) {
              int8_t intreturnedvalue;
              dxl_comm_result = packetHandler->read1ByteTxRx(portHandler, PacketReceivedInstDetail_1, PacketReceivedRegAddressValue, (uint8_t*)&intreturnedvalue, &dxl_error);

            } else if intByteCount == 2) {
              
              int16_t intreturnedvalue;
              dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, PacketReceivedInstDetail_1, PacketReceivedRegAddressValue, (uint16_t*)&intreturnedvalue, &dxl_error);
              
            } else if intByteCount == 4) {
              
              int32_t intreturnedvalue;
              dxl_comm_result = packetHandler->read4ByteTxRx(portHandler, PacketReceivedInstDetail_1, PacketReceivedRegAddressValue, (uint32_t*)&intreturnedvalue, &dxl_error);
              
            }
`

While having this in our firmware is not a big deal. All the work on front end software to track and send over register byte counts is very troublesome. For our software we have to store CSV files for every family of servos and do data imports, and lookups, and passing over of the byte length. It was a lot of added plumbing code to manage to make read/write to registers work. 

When our firmware is done and released we will be releasing the serial protocol for communicating with it. Since we have no choice but to have to know the register byte length that means that we have to ask our clients and customers to also have to pass it to our firmware. 

It seems redundant to have to tell the function what the byte length is. The DynamixelSDK should know the byte length internally for it's own register tables. Our request is that a version of readTxRx and writeTxRx be added that does not request the packet length. I imagine then the returned value can always be an int 32 to handle all integer sizes up to 4 bytes. This would help simplify code for your users and third party partners.

readTxRx
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/cpp/cpp_packethandler/#cpp-protocol2packethandler

writeTxRx
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/cpp/cpp_packethandler/#cpp-protocol2packethandler

Thank You!

Matt

kijongGil, 2018-09-10:
Hi @mtrossen :)
Thank you for your good feedback. I understand what you want.
But, the DynamixelSDK is basic tool for how to use dynamixel. It's hard to add a function to DynamixelSDK that checks what the byte length is.
Actually, we have released dynamixel-workbench(https://github.com/ROBOTIS-GIT/dynamixel-workbench) using DynamixelSDK a library. The dynamixel-workbench checks the byte length separately from DynamixelSDK. 
I hope that this will be help you.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp

Thanks,
Gilbert

mtrossen, 2018-10-09:
Hi Gilbert,
My apologies for the late reply. I wasn't getting notices that you had replied. Our firmware is written in arduino C++ and loaded onto the CM9.04. It appears that the workbench is a command line tool unless I am not understanding something? So it would not be of help for writing firmware it seems?

kijongGil, 2018-10-16:
Hi @mtrossen :)
Yes, you can write firmware using workbench into the CM9.04.
Maybe this [link](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/) will be help you.

Thanks,
Gilbert

mtrossen, 2018-10-16:
Hi Gilbert,

Thanks for the reply. I did see that there were workbench examples in the arduino examples folder and see now that it is available in the arduino IDE. We ended up writing our own function to handle looking up the register byte sizes in our firmware. Thanks for the kind help.

closed 2018-10-16 by mtrossen