ROBOTIS-GIT/DynamixelSDK #234: `[3.5.2][C++][USB2Dynamixel2][L54-50-S500-R Issue : My motor doesn't work`

lijing94123, 2018-09-09:
- I'm using `DynamixelSDK ver. 3.5.2
- I'm using `C++` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `L54-50-S500-R` and 'M54-60-S250-R'
- and I'm having an `issue` 
which is the torque of these motors can be enabled but the goal position cannot be written to PRO series motors.
And my basic algorithm is after initiating necessary packages, opening port and setting baudrate, I firstly set the goal moving speed and then enable the torque and then write the goal position, and disable the torque, close port. These steps for one motor can be found in figures attatached. The other one is just the same with only ID numbers changed.
I have checked both motors with teh software Dynamixel wizard and they work just fine. I am wondering what really happens here.
Hope I could get help here!

Thank you so much!
[image]
[image]
[image]
[image]

kijongGil, 2018-09-10:
Hi @lijing94123 :)
Did you check DXL control tables? Refer to follow link.
http://emanual.robotis.com/docs/en/dxl/pro/l54-50-s500-r/
http://emanual.robotis.com/docs/en/dxl/pro/m54-60-s250-r/

And then, try to run the **sync_read_write** example: http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_sync_read_write_protocol_2_0/#cpp-sync-read-write-protocol-20
If you get a error, please show me the error log.

Thanks,
Gilbert

lijing94123, 2018-09-10:
Thank you! 
I've checked the control table again, the goal position for PRO is 596 as listed in the control table. When I run my code, I can see it shows 'Dynamixel has been successfully connected' for both motors.
And I can directly run sync_read_write example without any error with my motors.

But I want different goal positions for these two motors, and I tried bulkread & bulkwrite in my code, but motors don't move. No error appears, just no motion happens.

kijongGil, 2018-09-10:
I modified sync_read_write code a bit.

#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>
#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Data Byte Length
#define LEN_PRO_GOAL_POSITION           4
#define LEN_PRO_PRESENT_POSITION        4

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL1_ID                         1                   // Dynamixel#1 ID: 1
#define DXL2_ID                         2                   // Dynamixel#2 ID: 2
#define BAUDRATE                        57600
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE_1    0             // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE_1    3500              // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)

#define DXL_MINIMUM_POSITION_VALUE_2    1000
#define DXL_MAXIMUM_POSITION_VALUE_2    1500

#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

int getch()
{
#if defined(__linux__) || defined(__APPLE__)
  struct termios oldt, newt;
  int ch;
  tcgetattr(STDIN_FILENO, &oldt);
  newt = oldt;
  newt.c_lflag &= ~(ICANON | ECHO);
  tcsetattr(STDIN_FILENO, TCSANOW, &newt);
  ch = getchar();
  tcsetattr(STDIN_FILENO, TCSANOW, &oldt);
  return ch;
#elif defined(_WIN32) || defined(_WIN64)
  return _getch();
#endif
}

int kbhit(void)
{
#if defined(__linux__) || defined(__APPLE__)
  struct termios oldt, newt;
  int ch;
  int oldf;

  tcgetattr(STDIN_FILENO, &oldt);
[... 199 lines cut ...]
    printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
  }
  else if ((dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION)) != 0)
  {
    printf("%s\n", getRxPacketError(PROTOCOL_VERSION, dxl_error));
  }

  // Disable Dynamixel#2 Torque
  write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL2_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE);
  if ((dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION)) != COMM_SUCCESS)
  {
    printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
  }
  else if ((dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION)) != 0)
  {
    printf("%s\n", getRxPacketError(PROTOCOL_VERSION, dxl_error));
  }

  // Close port
  closePort(port_num);

  return 0;
}

I hope that it will be help you.

lijing94123, 2018-09-12:
Thanks! I'll try.

kijongGil, 2018-09-17:
How are you doing? I hope everything is going great.

lijing94123, 2018-09-17:
So sorry for my late reply. 
I came up with issues as "identifier not found"... as marked with red lines in the picture attached, while the previous code didn't show these errors. And I wonder why these happen...

[image]
[image]

lijing94123, 2018-09-17:
I found another issue, which is, the present position and goal position of L54-50-S500-R are not corresponding to each other as shown in figure attached. And they turned to be the same when I enabled torque again, but the goal position turned to be the same as the present position instead of the motor moving to the goal position...
***********************************************************************************************************
[image]
***********************************************************************************************************
[image]

kijongGil, 2018-09-19:
Did you check this issue using R+ manager? Because you can not use R+ manager and DynamixelSDK at the same time. And I recommend `factory reset` using R+ manager.

lijing94123, 2018-09-24:
Hi, 
Thank you for your reply! I found that when I wrote some position for the L54 motor in R+ manager, its current has increased but no motion occurred. We believe that something's wrong with the motor. and we are trying to get the repair service.

Thank you!

robotpilot, 2018-09-30:
Hi, @lijing94123, I hope everything is going great. I want your problem to be resolved, and I will close this issue once. Please reopen or register a new issue at any time. We will support you. Thanks.

closed 2018-09-30 by robotpilot