ROBOTIS-GIT/DynamixelSDK #235: [3.5.4][C][U2D2][XM430-W210-R] Issue : Not receiving the encoder values`

karishmabava, 2018-09-10:
Hi,

My motors are running but I am getting [TxRxResult] There is no status packet! for all the motors. I am using Linux x86_64. All the motors are receiving the values and sending wrong/garbage values.

This is my read function

int read_func(int port_num, int pos, int id) {

     int index = 0;
     int dxl_comm_result = COMM_TX_FAIL;             // Communication result
     int dxl_goal_position[2] = { DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE };         // Goal position

     uint8_t dxl_error = 0;                          // Dynamixel error
     int32_t dxl_present_position = 0;               // Present position
     dxl_goal_position[index] = pos;

     dxl_present_position = read4ByteTxRx(port_num, PROTOCOL_VERSION, id, ADDR_PRO_PRESENT_POSITION);
     if ((dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION)) != COMM_SUCCESS) {
         printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
     }
     else if ((dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION)) != 0) {
         printf("%s\n", getRxPacketError(PROTOCOL_VERSION, dxl_error));
     }

     printf("[ID:%03d] GoalPos:%03d  PresPos:%03d\n", id, dxl_goal_position[index], dxl_present_position);

Thank you

kijongGil, 2018-09-11:
Hi @karishmabava :)
Did you update Dynamixel firmware?
I recommend using R+ manager for update firmware.
http://emanual.robotis.com/docs/en/software/rplus2/manager/

Thanks,
Gilbert

karishmabava, 2018-09-11:
Hi Gilbert,

I have an updated firmware and I am running on protocol 2.0

--Karishma

kijongGil, 2018-09-13:
Could you connect and move the Dynamixel by R+ manager?

karishmabava, 2018-09-13:
Hey got it... for some reason when I cloned the folder again for linux c... it started working....

Thanks!

closed 2018-09-13 by karishmabava