ROBOTIS-GIT/DynamixelSDK #236: [Master][C++][USB2Dynamixel2][MX64] Issue : Implementing function block

serogamon, 2018-09-12:
- I'm using `DynamixelSDK ver. Master

- I'm using `C++` Language

- I'm using `USB2Dynamixel2` serial converter

- I'm using MX64

- and I'm having an `issue` like what:

I have a Simulink modell with Image acquisition and processing. I would like to implement a function block that the position of the motor react to the changing signal. How can i do this?

At first i´ve tried to controll the Dynamixel with a constant signal for the position like shown in the photos.

[image]
[image]

But i get this Error:

[image]

kijongGil, 2018-09-13:
Hi @serogamon :)
I'm sorry that I don't understand your situation.
Could you give me the more information??

serogamon, 2018-09-17:
Hi, yes i´m able to controll the dynamixel with Matlab. I also tried to make my whole model with only Matlab, but here my problem is: i have to use a PID controller. In Simulink there is a PID block with an Input and an Output. When i use  the PID controller in Matlab i didn´t get a clear Output signal, only a transfer function. 
Maybe you know how to use this transfer function to controll the Dynamixel?

kijongGil, 2018-09-20:
I'm sorry, but as we said, we do not support Simulink. Later, I will test whether it works in Simulink.

robotpilot, 2018-09-30:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-09-30 by robotpilot

lijing94123, 2019-06-11:
> I'm sorry, but as we said, we do not support Simulink. Later, I will test whether it works in Simulink.

Hi, may I know whether you've tested dynamixel motors in simulink and does it work?

Thank you!