ROBOTIS-GIT/DynamixelSDK #241: [3.5.4][python][USB2Dynamixel][mx64AR] Cannot build linux32 library

clarelywork, 2018-09-21:
Hello!

I am trying to connect a robot to raspiberry pi Model B through USB port, I have the v3.5.4 library on the raspi.
I have the script in Python, but the linux system is changed from linux64 to linux32.

Since I moved the script from a linux64 system, I cleared the makefile for linux64:

[image]

then I went to linux32 folder and run:
$ make
and this is the result:

[image]

Can you let me know how this can be solved?

Thank you!

clarelywork, 2018-09-25:
Hello!

Someone from Robotis pointed out that if the date/time is not correct it might trow an error to the Makefile, so I corrected the time manually to match within 1 min error.
However, when I re-run the script, the error still exists:

[image]

Please let me know how this can be resolved, thank you!

Clare

ROBOTIS-zerom, 2018-09-28:
'-m32' tells the compiler to use and create 32bit objects, in contrast to '-m64' which would use/compile 64bit objects, and since the Pis ARM CPU only supports 32bit mode, both flags were left out.

just remove the '-m32' flag from Makefile and try again.

closed 2018-10-17 by gpwhd07