ROBOTIS-GIT/DynamixelSDK #243: [3.5.2][MATLAB][USB2Dynamixel2][XL-320] Issue: Motors behave erratically when I use 20 motors in syncrwite mode

dmrobotix, 2018-10-01:
I have 20 XL-320 split across two power cables. I use the groupSync set of methods to write position to the motors following the example code in sync_read_write.m for protocol 2 motors. Everything seems to work fine when I add up to 19 goal positions using groupSyncWriteAddparam but when I add in the 20th motor, the motors no longer move predictably. Is there a limitation on the number of motors that groupSync can handle? Thanks.

kijongGil, 2018-10-04:
Hi @dmrobotix :)
There is no limit on the number of motors in the DynamixelSDK.
But, If buadrate is too low or latency is too high, there is a possibility of problems when controlling multiple motors.
So, I recommend using a baudrate of 1M or 2M.

Thanks,
Gilbert.

kijongGil, 2018-10-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-10-16 by kijongGil

dmrobotix, 2018-10-18:
Thanks, I'm using 1M, but I'll try 2M.