ROBOTIS-GIT/DynamixelSDK #244: [3.6.1][C++/Arduino][OpenCR][XM430-W210-R] Incorrect Status Packet Error when using SyncRead

eng-Sund, 2018-10-01:
Hi,

I am getting an incorrect status packet error when I try to use sync read. The example code (modified for my motors) works but I am using the same relevant lines of code in my program and it doesn't work.

Here is the example code:

`#include <DynamixelSDK.h>

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Data Byte Length
#define LEN_PRO_GOAL_POSITION           4
#define LEN_PRO_PRESENT_POSITION        4

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL1_ID                         1                   // Dynamixel#1 ID: 1
#define DXL2_ID                         13                   // Dynamixel#2 ID: 2
#define BAUDRATE                        57600
#define DEVICENAME                      "3"                 // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      1600              // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      1800              // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

#define CMD_SERIAL                      Serial

int getch()
{
  while(1)
  {
    if( CMD_SERIAL.available() > 0 )
    {
      break;
    }
  }

  return CMD_SERIAL.read();
}

int kbhit(void)
{
  return CMD_SERIAL.available();
}

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  Serial.println("Start..");

   // Initialize PortHandler instance
  // Set the port path
  // Get methods and members of PortHandlerLinux or PortHandlerWindows
  dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

  // Initialize PacketHandler instance
  // Set the protocol version
  // Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
[... 415 lines cut ...]
    // Get Left Front Knee Dynamixel present position value
    pres_pos[2] = groupSyncRead.getData(dyn_id[2], pres_pos_ad, pres_pos_len);

    // Get Left Front Ankle Dynamixel present position value
    pres_pos[3] = groupSyncRead.getData(dyn_id[3], pres_pos_ad, pres_pos_len);

    Serial.print("ID:"); Serial.println(dyn_id[0]);
    Serial.print("  PresPos:"); Serial.println(pres_pos[0]);
    Serial.print("ID:"); Serial.println(dyn_id[1]);
    Serial.print("  PresPos:"); Serial.println(pres_pos[1]);
    Serial.print("ID:"); Serial.println(dyn_id[2]);
    Serial.print("  PresPos:"); Serial.println(pres_pos[2]);
    Serial.print("ID:"); Serial.println(dyn_id[3]);
    Serial.print("  PresPos:"); Serial.println(pres_pos[3]);
    Serial.println(" ");
  }

  // Close port
  portHandler->closePort();

}

void loop()
{
}`

eng-Sund, 2018-10-02:
Hi

I changed the number of motors being added as parameter from 16 to 10, it works!

```
for(int i=0;i<16;i++)
   {

    dxl_addparam_result = groupSyncRead.addParam(dyn_id[i]);
    if (dxl_addparam_result != true)
    {
      Serial.print("ID:"); Serial.print(dyn_id[i]); Serial.println("groupSyncRead addparam failed");
      return;
    }
  }
```

So when I change the '16' to '10' it works! but anything beyond 10 does not!! Is that a limit on the number of dynamixels? How do I get around the problem?

kijongGil, 2018-10-04:
Hi @eng-Sund :)
There is no limit on the number of motors in the DynamixelSDK.
Could you check your DXL's ID? If duplicate ID exists, it will not be executed.

Thanks,
Gilbert

eng-Sund, 2018-10-04:
Hi @kijongGil ,

Thanks for the reply! The ID's are unique as I am using them to do other stuff. Currently am using the readbyte function and it works fine, but there is pause in between my motions which I think is because I use readbyte to get motor position on at a time (16 motors) rather than using group sync read. Also, I am able to use group sync write just not group sync. The problem only occurs with group sync read.

Best,
Prasanna

kijongGil, 2018-10-05:
How about change baudrate to 1M or 2M?
A high baud rate is recommended when using multiple motors.

eng-Sund, 2018-10-05:
Hi 
I changed the baudrate to 1M. Now I get the error [TxRxResult] There is no status packet! if I try to do anything such as enabling torque, setting velocities etc. But everything (except group sync read) works at 57600 baud rate.

kijongGil, 2018-10-16:
I'm sorry for late reply.
Maybe baudrate of your DXL's is set 57600. You can change baudrate using r+ [manager](http://emanual.robotis.com/docs/en/software/rplus2/manager/) or [dynamixel_workbench](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/).

eng-Sund, 2018-10-16:
So the baud rate change works in the r+ manager (U2D2) but not in code (OpenCR)

kijongGil, 2018-10-17:
Could you tell me about your work environment? What did you use OS, IDE, language??

eng-Sund, 2018-10-18:
Windows 10. Arduino IDE, C++

eng-Sund, 2018-10-22:
I am using the following code:

```
if (portHandler->setBaudRate(BAUDRATE))
  {
    Serial.print("Succeeded to change the baudrate!\n");
  }
  else
  {
    Serial.print("Failed to change the baudrate!\n");
    return;
  }
```

eng-Sund, 2018-10-23:
Hi,

So, I was able to make the dynamixels work at 2M baudrate. I had change the baudrate in r+ Manager and in the OPEN CR code. I still can't make the sync read work. It gives me incorrect status packet error.

Also, what is the baud rate of the Open CR?

kijongGil, 2018-10-24:
The baud rate of the OpenCR is 57600. Incorrect status the packet error is most likely to occur when the protocol is wrong or when the wrong control table is used.
I tested your code, It is work. So, It is not control table problem.

What did you use protocol version?

eng-Sund, 2018-10-24:
Sorry, I meant what is the max baudrate of the Open CR?

I am using Ver 2.0

closed 2018-10-30 by gpwhd07