ROBOTIS-GIT/DynamixelSDK #245: [3.5.2][C++][USB2Dynamixel2][MX106R][L54-50-S500R][L54-50-S500R] Issue : Cannot combine MX & PRO series motors together

lijing94123, 2018-10-05:
Hi, I have an issue with Dynamixel motors when combining MX and PRO series motors together.

[sync_read_write.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2450522/sync_read_write.txt)
[bulk_read.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2450523/bulk_read.txt)

- I'm using `DynamixelSDK ver. 3.5.2`

- I'm using `C++` Language

- I'm using `USB2Dynamixel2` serial converter

- I'm using four `MX106R` , one `L54-50-S500R` and one `M54-60-S250-R`

- and I'm having an `issue`  with combining these 6 motors together. 
I would like to set the speed and goal position for these motors and in my code I only set goal speed and position for MX106R motors and when I tried to set goal position for PRO motors, they couldn't move. The code I used to driver these motors was modified based on protocol_combined example and sync_read_write for PRO motors.
In details, I have checked the dynamixel wizard afterwards, and I found the goal position seemed to be changed but the motors are not in the right position. And if I simply ran code as sync_read_write in protocol2 for PRO motors, they performed just fine.

I have attached these two codes I have run. Could you please help me check whether there is any issue? 

Thank you so much for your help!

lijing94123, 2018-10-06:
I have one more thing to add. I can only detect PRO series motor in dynamixel wizard when I connected all the motors together with appropriate power supplys. And I can detect MX motors in dynamixel wizard when I only connect MX motors to PC.
But when I tried to drive all these 3 motors by c++ combined protocol, MX motors worked well.

lijing94123, 2018-10-07:
Hi, sorry I have one more inssue. 

I've noticed that it is recommended to have the same protocol for multiple motors so I have tested MX motors with protocol 2.0 in MATLAB (version 3.5.2, read_write example for one motor) but it shows that there is no status packet.

As you can see in the picture attached.  And I have attched the code below.

I am wondering whether it is because that my MX 106R firmware needs to be updated for using protocol 2.0 and how can I do it? Or are there any other causes?
[image]

`clc;
clear all;

lib_name = '';

if strcmp(computer, 'PCWIN')
  lib_name = 'dxl_x86_c';
elseif strcmp(computer, 'PCWIN64')
  lib_name = 'dxl_x64_c';
elseif strcmp(computer, 'GLNX86')
  lib_name = 'libdxl_x86_c';
elseif strcmp(computer, 'GLNXA64')
  lib_name = 'libdxl_x64_c';
elseif strcmp(computer, 'MACI64')
  lib_name = 'libdxl_mac_c';
end

% Load Libraries
if ~libisloaded(lib_name)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h', 'addheader', 'packet_handler.h');
end

% Control table address
ADDR_MX_TORQUE_ENABLE       = 64;         % Control table address is different in Dynamixel model
ADDR_MX_GOAL_POSITION       = 116;
ADDR_MX_PRESENT_POSITION    = 132;

% Protocol version
PROTOCOL_VERSION            = 2.0;          % See which protocol version is used in the Dynamixel

% Default setting
DXL_ID                      = 3;            % Dynamixel ID: 1
BAUDRATE                    = 57600;
DEVICENAME                  = 'COM15';      

TORQUE_ENABLE               = 1;            % Value for enabling the torque
TORQUE_DISABLE              = 0;            % Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 1000;      % Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 3000;       % and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
DXL_MOVING_STATUS_THRESHOLD = 20;           % Dynamixel moving status threshold

ESC_CHARACTER               = 'e';          % Key for escaping loop

COMM_SUCCESS                = 0;            % Communication Success result value
COMM_TX_FAIL                = -1001;        % Communication Tx Failed

% Initialize PortHandler Structs
% Set the port path
% Get methods and members of PortHandlerLinux or PortHandlerWindows
port_num = portHandler(DEVICENAME);

% Initialize PacketHandler Structs
packetHandler();

index = 1;
dxl_comm_result = COMM_TX_FAIL;           % Communication result
dxl_goal_position = [DXL_MINIMUM_POSITION_VALUE DXL_MAXIMUM_POSITION_VALUE];         % Goal position

dxl_error = 0;                              % Dynamixel error
dxl_present_position = 0;                   % Present position

% Open port
if (openPort(port_num))
    fprintf('Succeeded to open the port!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to open the port!\n');
    input('Press any key to terminate...\n');
    return;
end

% Set port baudrate
if (setBaudRate(port_num, BAUDRATE))
    fprintf('Succeeded to change the baudrate!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to change the baudrate!\n');
    input('Press any key to terminate...\n');
    return;
end

% Enable Dynamixel Torque
write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_ENABLE);
dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
if dxl_comm_result ~= COMM_SUCCESS
    fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
elseif dxl_error ~= 0
    fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
else
    fprintf('Dynamixel has been successfully connected \n');
end

while 1
    if input('Press any key to continue! (or input e to quit!)\n', 's') == ESC_CHARACTER
        break;
    end

    % Write goal position
    write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_GOAL_POSITION, typecast(int32(dxl_goal_position(index)), 'uint32'));
    dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
    dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
    if dxl_comm_result ~= COMM_SUCCESS
        fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
    elseif dxl_error ~= 0
        fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
    end

    while 1
        % Read present position
        dxl_present_position = read4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_PRESENT_POSITION);
        dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
        dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
        if dxl_comm_result ~= COMM_SUCCESS
            fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
        elseif dxl_error ~= 0
            fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
        end

        fprintf('[ID:%03d] GoalPos:%03d  PresPos:%03d\n', DXL_ID, dxl_goal_position(index), typecast(uint32(dxl_present_position), 'int32'));

        if ~(abs(dxl_goal_position(index) - typecast(uint32(dxl_present_position), 'int32')) > DXL_MOVING_STATUS_THRESHOLD)
            break;
        end
    end

    % Change goal position
    if index == 1
        index = 2;
    else
        index = 1;
    end
end

% Disable Dynamixel Torque
write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE);
dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
if dxl_comm_result ~= COMM_SUCCESS
    fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
elseif dxl_error ~= 0
    fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
end

% Close port
closePort(port_num);

% Unload Library
unloadlibrary(lib_name);

close all;
clear all;
`

lijing94123, 2018-10-14:
Hi, I've tried to update the firmware of my MX106R to version 40, but still, I cannot use protocol 2.0 for MX106R. What could be the issue? Do I need to update my protocol package?

kijongGil, 2018-10-16:
Hi @lijing94123 :)
I'm sorry for late reply.
First, Check your the values in the control table of DXLs(protocol, ID) by R+ manager.
refer to this link: http://emanual.robotis.com/docs/en/software/rplus2/manager/

And then, If you want to move several DXL, make sure each DXL's control table. 
For example, MX106's control table is [this](http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/#control-table-data-address), Pro's control table is [this](http://emanual.robotis.com/docs/en/dxl/pro/m54-40-s250-r/#control-table-data-address).

Thanks,
Gilbert

lijing94123, 2018-10-16:
Thanks! I can use the MX106R for protocol 2.0 now. And I've double checked the control table. I can also actuated all 6 motors together now but I have some issues of real time control. 
I simply write goal positions for each motors and the goal positions are calculated by other functions and determined by input data with a refresh rate around 1 kHz. But I found that the motors only move when I inserting "read and print present position of motors" after writing goal positions and before disable their torque. They won't move when I simplt write the goal position and disable their torque afterwards. You can see the attched txt files for more information of my code.
It seems like the motors have been disabled torque even before they reach their goal positions. But if I inserted the "read&print present position", the whole process would become very slow. And I don't want to speed up the motors velocities.
Do you have any suggestions?
Thank you!

[read_write_combined.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2482679/read_write_combined.txt)

kijongGil, 2018-10-17:
I think you need to add groupSync function.
Refer to [this](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_sync_read_write_protocol_2_0/#cpp-sync-read-write-protocol-20)

For example,
  // Add parameter storage for Dynamixel#1 present position value
  dxl_addparam_result = groupSyncRead.addParam(DXL1_ID);

lijing94123, 2018-10-18:
Hi, thank you! But groupSync can't be used with both PRO and MX motors as I tested. I've double checked that with only MX106R motors or only PRO motors, it works. And I don't think groupSync can speed up the position control process.

It seems like my motors cannot catch up with input data refresh rate. And I've tried to not to enable and disable torque of motors each time I send data to trigger motions. I enabled motors torque at the very beginning and disabled the torque at the end, sending data to change goal positions are in between whenever I pushed the button. And I decreased the refresh date of input data to 500Hz instead of 1KHz. 

But motors still cannot move when I don't insert "read/print present position".

Do you have any idea of how to speed up the whole control process?

lijing94123, 2018-10-18:
I was suggested to change to velocity control instead of position control. And may I have an example of velocity control of PRO motors in protocol 2.0

Thank you!

kijongGil, 2018-10-18:
I think this problem is that incorrectly byte size. In your code, you used write4ByteTxRx function for goal position. But, MX's goal position is 2 Byte.Maybe this caused an overflow.

Have you ever controlled only one MX-106? If not, I recommend controlling only one motor.

lijing94123, 2018-10-18:
But I am using protocol 2.0 for MX106R where the goal position is 4 bytes as listed [http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/#control-table-of-eeprom-area]

lijing94123, 2018-10-18:
[MX106R protocol 2.0](http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/#control-table-of-eeprom-area)

kijongGil, 2018-10-18:
It's my bad, I confused protocol 1.0.
I tested control to together MX-106 and XM430. It's working well.
As mentioned earlier, check your DXL's firmware(latest version: v42), ID and baudrate.
Then, If you want to control 6 DXL, you need 6 packetHandler.

```
dxl_comm_result, dxl_error = packetHandler_1.write4ByteTxRx(portHandler, DXL_ID_1, ADDR_PRO_GOAL_POSITION, dxl_goal_position[index])
dxl_comm_result, dxl_error = packetHandler_2.write4ByteTxRx(portHandler, DXL_ID_2, ADDR_PRO_GOAL_POSITION, dxl_goal_position[index])
```

lijing94123, 2018-10-18:
I do find an interesting thing.
I calculated the motor angle in function 'int p1()', which will return an integer working as the goal position for my motor and I set it within the motion range. And I directly pass this value to the goal_position of my motor by "int dxl1_goal_position = p1()", but I found that p1() and dxl1_goal_position return different values when I print them out.
How come this happened?
And am I wrong about the byte size of goal position of MX106R in protocol 2.0?

lijing94123, 2018-10-18:
I need 6 DXL for each even four of them are all MX106R? Could you please explain the logic? I am so confused.
DXL firmwares of my MX106R have been updated to version 42, IDs are right and so is the baudrate. There is no issue with connecting.
Thank you for your help!

kijongGil, 2018-10-18:
This is my test python code. but you can refer to it enough.
This code is not perfect, but it may be appropriate to confirm that you are moving both types of motors.
```
import os

if os.name == 'nt':
    import msvcrt
    def getch():
        return msvcrt.getch().decode()
else:
    import sys, tty, termios
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)
    def getch():
        try:
            tty.setraw(sys.stdin.fileno())
            ch = sys.stdin.read(1)
        finally:
            termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)
        return ch

from dynamixel_sdk import *                    # Uses Dynamixel SDK library

# Control table address
ADDR_PRO_TORQUE_ENABLE      = 64               # Control table address is different in Dynamixel model
ADDR_PRO_GOAL_POSITION      = 116
ADDR_PRO_PRESENT_POSITION   = 132

# Protocol version
PROTOCOL_VERSION            = 2.0               # See which protocol version is used in the Dynamixel

# Default setting
DXL_ID_1                    = 1
DXL_ID_2                    = 2                 # Dynamixel ID : 1
BAUDRATE                    = 57600             # Dynamixel default baudrate : 57600
DEVICENAME                  = '/dev/ttyUSB0'    # Check which port is being used on your controller
                                                # ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

TORQUE_ENABLE               = 1                 # Value for enabling the torque
TORQUE_DISABLE              = 0                 # Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 10           # Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 4000            # and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
DXL_MOVING_STATUS_THRESHOLD = 20                # Dynamixel moving status threshold

index = 0
dxl_goal_position = [DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE]         # Goal position

# Initialize PortHandler instance
# Set the port path
# Get methods and members of PortHandlerLinux or PortHandlerWindows
portHandler = PortHandler(DEVICENAME)

# Initialize PacketHandler instance
# Set the protocol version
# Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
packetHandler = PacketHandler(PROTOCOL_VERSION)
packetHandler_2 = PacketHandler(PROTOCOL_VERSION)
# Open port
if portHandler.openPort():
    print("Succeeded to open the port")
else:
    print("Failed to open the port")
    print("Press any key to terminate...")
    getch()
    quit()

# Set port baudrate
if portHandler.setBaudRate(BAUDRATE):
    print("Succeeded to change the baudrate")
else:
[... 55 lines cut ...]
            break

    # Change goal position
    if index == 0:
        index = 1
    else:
        index = 0

# Disable Dynamixel Torque
dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL_ID_1, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))

dxl_comm_result, dxl_error = packetHandler_2.write1ByteTxRx(portHandler, DXL_ID_2, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler_2.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler_2.getRxPacketError(dxl_error))

# Close port
portHandler.closePort()

```

lijing94123, 2018-10-18:
Hi, thank you so much! And I am so sorry for my so many questions and ambiguous explanations.

Actually I can control 6 motors together if I simply make them change goal position between two values. 

But I cannot make them move to the right position at the given velocity when I keep sending instructions to change goal positions. They just didn't move.

And I think I've given the instructions of changing goal positions too fast (but it is already the lowest rate I could use for real-time control) which has 'confused' motors. Another reason could be that the given velocity and the given goal position of the motors cannot be realized the same time since the acceleration exceeds its range.(? just a guess) So they just stopped there.

And since the motors need to generate velocity trajectory and acceleration trajectory after they received position control instructions (if I understood the documention correctly), I think directly velocity control would be better in this case.

And I set abs(goal_position-present_position)*0.1 to be the goal velocity, but then I found that the goal_position is not what I passed to it....(this doesn't happen before...could it be a data type issues?)

So I would be appreciated that if you could give me an example of velocity control of two different types of motors. And please let me know whether I misunderstood any of your documentations as I've mentioned above and let me know your idea with my issues or questions about my explanations.

Thank you so much!

kijongGil, 2018-10-23:
I'm sorry for late reply.
I thought your question was simply to move the motors of the other series at the same time. But if you want a little more sophisticated control, refer to [OpenManipulator](http://emanual.robotis.com/docs/en/platform/openmanipulator/#software-setup) and [github](https://github.com/ROBOTIS-GIT/open_manipulator).
I think it will be help you.

Thanks.

closed 2018-10-30 by kijongGil