ROBOTIS-GIT/DynamixelSDK #246: MX-12W servo: [TxRxResult] There is no status packet!

kevinpei2001, 2018-10-08:
ISSUE TEMPLATE ver. 1.1.1

protocol 1.0

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

[3.6.0]

- I'm using `C++` Language

[Python] 

- I'm using `USB2Dynamixel2` serial converter

[U2D2]

- I'm using `XM430-W210-R`

MX-12W

- and I'm having an `issue` like what

I set up the enviroment: python setup.py install
I connected the hardware accordingly based on:
http://emanual.robotis.com/docs/en/parts/interface/u2d2/

When run python read_write.py, it gave:

[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)

I run ping.py it gave the same issue. However, when I changed the BAUDRATE to 1000000 the ping.py worked. I changed the same value in read_write.py it didn't work.

I checked your web link:
http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-12w.htm
and tried every baudrate mentioned there. There was no luck. Some of them showed Failed to change BAUDRATE; the others showed [TxRxResult] There is no status packet!

So tricky. Am I missing something?

Kevin

kevinpei2001, 2018-10-08:
When I run ping.py with 1000000 baudrate it gave this:

[root@artik protocol1_0]# python ping.py
Succeeded to open the port
Succeeded to change the baudrate
[ID:001] ping Succeeded. Dynamixel model number : 360

kevinpei2001, 2018-10-08:
Hi, I used 1000000 baudrate in read_write.py. The no status packet error disappeared. However, it gave:

Succeeded to open the port
Succeeded to change the baudrate
Dynamixel has been successfully connected
Press any key to continue! (or press ESC to quit!)
[RxPacketError] Out of range error!
[ID:001] GoalPos:000  PresPos:000
Press any key to continue! (or press ESC to quit!)

Here is the configuration:

# Control table address
ADDR_MX_TORQUE_ENABLE      = 24               # Control table address is different in Dynamixel model
ADDR_MX_GOAL_POSITION      = 116
ADDR_MX_PRESENT_POSITION   = 132

# Protocol version
PROTOCOL_VERSION            = 1.0               # See which protocol version is used in the Dynamixel

# Default setting
DXL_ID                      = 1                 # Dynamixel ID : 1
BAUDRATE                    = 1000000             # Dynamixel default baudrate : 57600
DEVICENAME                  = '/dev/ttyUSB0'    # Check which port is being used on your controller
                                                # ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

TORQUE_ENABLE               = 1                 # Value for enabling the torque
TORQUE_DISABLE              = 0                 # Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 0           # Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 4000            # and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)

kijongGil, 2018-10-15:
Hi @kevinpei2001 :)
First, I'm sorry for late reply. I participated in the exhibition last week.
Could you update DXL's firmware? Refer to this link: http://emanual.robotis.com/docs/en/software/rplus2/manager/

And then what do you use protocol version?
If you want to use protocol version 1.0, your DXL's has to set protocol version 1.0 by R-manager or else.
Please check your protocol version.

Thanks,
Gilbert