ROBOTIS-GIT/DynamixelSDK #249: [0.0.1][Arduino][OpenCM9.04/OpenCR1.0] Issue how to read information from multiple servo for more than one field

KurtE, 2018-10-12:
Not sure if best to ask this here or Ask it on OpenCM9.04 and/or OpenCR project. 

Regardless it is not tied to any specific release or the like.  I am currently using develop branch on OpenCR as my latest changes were folded in there and on PR against develop branch on OpenCM9.04.

Also some of this applies to Dynamixel Workbench as well. 

Issue came up in the forum posting: http://en.robotis.com/service/forum_view.php?bbs_no=2407000&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

Issue: suppose I have a robot that has 18XL430-W250  servos and I would like to read in status of all 18 servos for multiple fields (example Present Load, Present Velocity and Present Position).  How do I do this using the SDK as Quickly as possible?  

In my own library code in the past I would do this using a Sync Read, as I can start reading at address 126 and read in 10 bytes for each servo.  

So I would hope I could create a group sync read item like:
```
  dynamixel::GroupSyncRead groupSyncRead(portHandler, packetHandler, 126, 10);
...
 dxl_addparam_result = groupSyncRead.addParam(DXL1_ID);
// and all of the other IDS... 
```

Then when I do the read I would hope I could just do: 
```
dxl_comm_result = groupSyncRead.txRxPacket();
if (dxl_comm_result != COMM_SUCCESS) packetHandler->getTxRxResult(dxl_comm_result);
...
dxl1_present_position = groupSyncRead.getData(DXL1_ID, 126, 10);
...
```
And I think everything would work up to the getData call.  The issue is getData can only handle data lengths of 1, 2, or 4 bytes. 

So Propose add another member method:  Maybe that would support doing something like:
```
uint8_t read_buffer[10];
dxl_comm_result  = groupSyncRead.getData(DXL1_ID, 126, 10, read_buffer);
if (dxl_comm_result != COMM_SUCCESS)) {
    uint16_t pres_current = read_buffer[0] + ((uint16_t)read_buffer[1] << 8);

    uint32_t pres_vel = read_buffer[2] + ((uint32_t)read_buffer[3] << 8) + ((uint32_t)read_buffer[4] << 16) + ((uint32_t)read_buffer[5] << 24);

    uint32_t pres_pos = read_buffer[6] + ((uint32_t)read_buffer[7] << 8) + ((uint32_t)read_buffer[8] << 16) + ((uint32_t)read_buffer[9] << 24);
```

Note Bulk Read has the same issue and can be solved the same way.

KurtE, 2018-10-12:
Actually I think I can answer it myself.  I wondered why I need to pass in address and length to the getData Call... But I think that is maybe the answer...

That is I should be able to change the above to be something like:
```
dxl_comm_result = groupSyncRead.txRxPacket();
if (dxl_comm_result != COMM_SUCCESS) packetHandler->getTxRxResult(dxl_comm_result);
...
 uint16_t pres_current = groupSyncRead.getData(DXL1_ID, 126, 2);
 uint32_t pres_vel =  groupSyncRead.getData(DXL1_ID, 128, 4);
 uint32_t pres_pos =  groupSyncRead.getData(DXL1_ID, 132, 4);
```
Actually can and maybe should just close this issue out, except maybe would be nice to have some documentation on this and/or example that shows this.

Sorry for maybe false call

kijongGil, 2018-10-17:
Hi @KurtE :)
I'm sorry for late reply. We discussed how to manage the dynamixelSDK. Currently there is a difference between the dynamixelSDK used for OpenCR and the official dynamixelSDK from [github](https://github.com/ROBOTIS-GIT/DynamixelSDK). 
It's hard to sync two dynamixelSDKs perfectly, so far I've managed to use at least the same function names. So, When you request PR to DynamixelSDK in OpenCR. I will confirm and update official dynamixelSDK.

Thanks,
Gilbert

closed 2019-02-27 by gpwhd07