ROBOTIS-GIT/DynamixelSDK #256: [3.6.2][MATLAB][USB2Dynamixel][MX-28AT]/[MX_64AT] SyncRead Issue

xzhang88, 2018-10-31:
I'm using sync read to get the position for 6 servos (3 x MX-28AT and 3 x MX-64AT) but the results are wrong/mixed randomly when I fetch data. Most of the time its fine but every 4th or 5th fetch the positions are wrong.

Running protocol 2, baud rate is at 1M for all the servos. Matlab is through Ubuntu and I also varied the serial port latency timer between 1/2/4/8/16ms and errors still occur randomly. Sequential read4byte is fine without error but is too slow for the setup I have when polling 6 servos sequentially. 

Pseudo code for the syncRead:
1. Initialize sync read structure
groupread_num = groupSyncRead(port_num, 2, ADDR_MX, LEN_MX);
2. Add parameters for 6 servos
dxl_addparam_result = groupSyncReadAddParam(groupread_num, DXL_num);  [repeated for 6 servo IDs]
3. Fetch data
groupSyncReadTxRxPacket(groupread_num);
4. Unpack data for each servo
Data(i) = groupSyncReadGetData(groupread_num, DXL_num, ADDR_MX,LEN_MX); [repeated for 6 servo IDs]
5. Repeat from 3.

xzhang88, 2018-11-27:
Any updates to this?

kijongGil, 2019-01-29:
I'm sorry for late reply.
we set to latency time 1ms and used 6 DOF manipulator. It is working well.
Cloud you tell me what you get errors?

xzhang88, 2019-01-30:
I get all zeros on the servos when I unpack the data when I syncread 6 servos. I actually have 36 servos total and the error occurs much more often with that many.

kijongGil, 2019-01-31:
I think that you repeat 6 times fetch data and unpack data in a loop, right?
You have to call groupSyncReadTxRxPacket(groupread_num) one time in a loop.

I mean that maybe you call to groupSyncReadTxRxPacket(groupread_num) -> 
groupSyncReadGetData(groupread_num, DXL_1, ADDR_MX,LEN_MX) -> groupSyncReadTxRxPacket(groupread_num) -> groupSyncReadGetData(groupread_num, DXL_2, ADDR_MX,LEN_MX) -> more...

Try to this groupSyncReadTxRxPacket(groupread_num) -> groupSyncReadGetData(groupread_num, DXL_1, ADDR_MX,LEN_MX) -> groupSyncReadGetData(groupread_num, DXL_2, ADDR_MX,LEN_MX) -> more...

xzhang88, 2019-01-31:
For each read instance, I only fetch data once using groupSyncReadTxRxPacket then groupSyncReadGetData 6 times to get data for each servo. The problem is that some of these instances after fetching gives all 0's when I use groupSyncReadGetData

kijongGil, 2019-02-01:
How did you set up latency time? If you change serial port latency timer, you should change to latency time in DynamixelSDK. Please refer to this [code](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/05dcc5c551598b4d323bf1fb4b9d1ee03ad1dfd9/c/src/dynamixel_sdk/port_handler_linux.c#L34).

closed 2019-02-27 by gpwhd07