ROBOTIS-GIT/DynamixelSDK #257: Dynamixel using C++

hamidnrn, 2018-11-09:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.6.2, I'm using `XM430-W210-R`,I have an issue to run the motors ,i"m using visual studio ,when I debuge it it keep giving me the following error  in windows console:

Succeeded to open the port !
[ PorthandlerWindows::Setupport]Error opening serial port !
Failed to open the port 
press any key to terminate ....

this is my code:::
 
/* Author: Ryu Woon Jung (Leon) */

//
// *********     multi_port Example      *********
//
//
// Available Dynamixel model on this example : All models using Protocol 2.0
// This example is tested with two Dynamixel PRO 54-200, and two USB2DYNAMIXEL
// Be sure that Dynamixel PRO properties are already set as %% ID : 1 / Baudnum : 1 (Baudrate : 57600)
//

#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>

#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL1_ID                         1                   // Dynamixel#1 ID: 1
#define DXL2_ID                         2                   // Dynamixel#2 ID: 2
#define BAUDRATE                        9600
#define DEVICENAME1                     "COM8"      // Check which port is being used on your controller
#define DEVICENAME2                     "COM8"      // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0             // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4000              // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

int getch()
{
#if defined(__linux__) || defined(__APPLE__)
	struct termios oldt, newt;
	int ch;
	tcgetattr(STDIN_FILENO, &oldt);
	newt = oldt;
	newt.c_lflag &= ~(ICANON | ECHO);
[... 221 lines cut ...]
	dxl_comm_result = packetHandler->write1ByteTxRx(portHandler2, DXL2_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
	if (dxl_comm_result != COMM_SUCCESS)
	{
		printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
	}
	else if (dxl_error != 0)
	{
		printf("%s\n", packetHandler->getRxPacketError(dxl_error));
	}

	// Close port1
	portHandler1->closePort();

	// Close port2
	portHandler2->closePort();

	return 0;
}

I'd be grateful if you can help me resolve this issue.
Thanks

  

-

kijongGil, 2018-11-23:
Hi @hamidnrn :)
I'm sorry for late reply. This is multi port example, So, you have to connect two port.
But, you defined DEVICENAME1 and DEVICENAME as same port(COM8).

Cloud you tell me about your H/w connection?

Thanks,
Gilbert

hamidnrn, 2018-11-26:
I need two dynamixels to work that's why i used multi port, is there any errors i made?could u plz tell me where?

kijongGil, 2018-11-26:
Because this example code you posted is multi-port example. Multi-port is suitable for managing the DXL by port. But, if you want to just move 2 or more DXL, multi-port is also possible, but it is better to use sync_read_write example.

hamidnrn, 2018-11-27:
thanks a million for ur response , Actually I faced issues using the multi_port so I'll try sync_read_write example ,hope it works for my two dynamixles

kijongGil, 2018-12-03:
How about your problem?

closed 2018-12-12 by gpwhd07