ROBOTIS-GIT/DynamixelSDK #259: Torque control using Dynamixel SDK python

lenonrobot, 2018-11-20:
Thanks for your python code of DynamixelSDK. Is there any python code using the torque control mode of XM430-W350-R motors? What I really need is to (controller) output joint torques of a two-joint robot arm, rather than joint positions.

Some torque control by c++:

http://wiki.ros.org/dynamixel_controllers/Tutorials/Creating%20a%20dual%20joint%20torque

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#tutorials

thanks if anyone could help me out. thanks again.

kijongGil, 2018-11-22:
Hi, xiaofeng :)

Unfortunately, there is no python code using torque control.

For getting joint torques, you have to use read2ByteTxRx and read present current of control table. refer to [control table](http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/). and then calculate torque from present current as like dynamixel_workbench.

Thanks,
Gilbert

lenonrobot, 2018-11-22:
Gilbert,
thanks for your quick reply. I will try what you suggested. I may let you know if there will be problems onto it. thanks again.

Best regards,
xiaofeng

kijongGil, 2018-11-23:
I hope your problem is solved. I'm going to close this issue. If you have an another problem, you can reopen it anytime. Thanks.

closed 2018-11-23 by kijongGil

spadetoone, 2019-12-15:
> Gilbert,
> thanks for your quick reply. I will try what you suggested. I may let you know if there will be problems onto it. thanks again.
> 
> Best regards,
> xiaofeng

xiaofeng
Do you have solve the problem that use python to torque control.

YechenZhou, 2023-03-25:
Do you have solve the problem that use python to torque control? 

> > Gilbert,
> > thanks for your quick reply. I will try what you suggested. I may let you know if there will be problems onto it. thanks again.
> > Best regards,
> > xiaofeng
> 
> xiaofeng Do you have solve the problem that use python to torque control.

 Do you have solve the problem that use python to torque control?

> > Gilbert,
> > thanks for your quick reply. I will try what you suggested. I may let you know if there will be problems onto it. thanks again.
> > Best regards,
> > xiaofeng
> 
> xiaofeng Do you have solve the problem that use python to torque control.

Have you solved the problem that use python to torque control? @spadetoone @lenonrobot