ROBOTIS-GIT/DynamixelSDK #261: currents and torques of xm430

lenonrobot, 2018-11-23:
Hi, I have questions:

(1) the relationship between currents and torques

In Fig. of 'Performance Graph', the relationship between currents (C) and torques(Tau) can be given by: C = A*Tau + B, how could I obtain the parameters A and B?

(2) Since the outputs of my controller are the motor torques, then could I use the equation C = A*Tau + B to control the motor current? is it right?

(3)  Could the range of the goal current (address:102) be [-1193, +1193]? Because the current limit (address:38) is 1193, is it right?

sorry for a list of questions. I donot know the control accuracy of the current mode. Previously, I tried to figure out the methods that can translate joint torques to positions, they vary over various robots. Please let me know if you know any general ways of translating joint torques to positions, it also fits to my controller.  thanks again.

kijongGil, 2018-11-27:
Hi, lenonrobot :)
I'm sorry for late reply. 
1) I don't know well about obtain these parameters. I can just advice how to get currents of DXL. 
2) Yes, I think you can control the DXL. 
3) Yes, you're right.

Unfortunately, I don't well about joint torques and position control. DynamixelSDK is just basic development tool for control. I recommend to refer [dynamixel-workbench](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#torque-controller) and [openmanipulator](http://emanual.robotis.com/docs/en/platform/openmanipulator/).

lenonrobot, 2018-11-27:
thanks all, I am figuring them out. thanks again.

kijongGil, 2018-12-03:
I'm going to close this issue. If you have an another problem, you can reopen it whenever.

closed 2018-12-03 by kijongGil