ROBOTIS-GIT/DynamixelSDK #262: [c][rs485][xm430-w350] how can I make a move?

gooinc2, 2018-11-29:
Hello,
- I am using XM430-W350 motor
- (Internally, I have a problem in receiving a packet. So I have been trying to send a packet only.) 
   It works for LED on/off but doesn't move at all for a goal position.
- I am based on http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/c_read_write_protocol_2_0/#c-protocol-20

- As I have a problem in receiving a packet, at this moment, I am just trying to switching between DXL_MINIMUM_POSITION_VALUE and DXL_MAXIMUM_POSITION_VALUE.
I did TORQUE_ENABLE prior to GOAL_POSITION.
Is there anything else for me to enable?

Thanks for your help.

[snippet of codes]
#define ADDR_PRO_TORQUE_ENABLE           64
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

#define PROTOCOL_VERSION                2.0

#define DXL_MINIMUM_POSITION_VALUE      -150000
#define DXL_MAXIMUM_POSITION_VALUE      150000

...

int dxl_goal_position[2] = { DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE };

  // Enable Dynamixel Torque
  write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_ENABLE);

  while (1)
  {
    printf("Press any key to continue! (or press ESC to quit!)\n");
    if (getch() == ESC_ASCII_VALUE)
        break;
	
       //Write goal position
    write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_GOAL_POSITION, dxl_goal_position[index]);
	
    ////do
    {
      // Read present position
      dxl_present_position = read4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_PRESENT_POSITION);
      if ((dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION)) != COMM_SUCCESS)
      {
        printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
      }
      else if ((dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION)) != 0)
      {
        printf("%s\n", getRxPacketError(PROTOCOL_VERSION, dxl_error));
      }

      printf("[ID:%03d] GoalPos:%03d  PresPos:%03d\n", DXL_ID, dxl_goal_position[index], dxl_present_position);

    } ////while ((abs(dxl_goal_position[index] - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD)); 

    // Change goal position
    if (index == 0)
    {
      index = 1;
    }
    else
    {
      index = 0;
    }
  }

gooinc2, 2018-11-29:
Hello,

it works with smaller values than the sample code per xm430-w350.
[sample code]
#define DXL_MINIMUM_POSITION_VALUE      150000 
#define DXL_MAXIMUM_POSITION_VALUE      150000
[my test]
#define DXL_MINIMUM_POSITION_VALUE      0 
#define DXL_MAXIMUM_POSITION_VALUE      1000

From xm430-w350 data, it looks that Initial Value is  0 ~ 4,095. 
1) Is this correct?
2.4.17 MAX/MIN Position Limit (48,52)
|Unit       | Value Range               |
|0.088 [°] | 0 ~ 4,095(1 rotation) |

2) what is 'Extended Position Control Mode'?
Mode                                                 | Values                                                                | Description
Position Control Mode                        | Min Position Limit(52) ~ Max Position Limit(48) | Initial Value : 0 ~ 4,095
Extended Position Control Mode         | -1,048,575 ~ 1,048,575                                    | -256[rev] ~ 256[rev]
Current-based Position Control Mode | -1,048,575 ~ 1,048,575                                    | -256[rev] ~ 256[rev]

Thanks.

kijongGil, 2018-11-30:
Hi, gooinc2 :)
Yes, the range of position of XM-430 is 0~4,095(1 rotation).
So, you made appropriate modifications to the code.
Extended Position control mode is that the dynamixel rotates more than one turn.
For example, If you set goal position to 10,000, dynamixel rotates 2 turn and has 1,810(10,000 - 2 * 4,095) position.

Thanks,
Gilbert

gooinc2, 2018-12-01:
Thanks Gilbert for your help.
One question. How am I able to make infinite revolutions forwards or backwards until I want to stop?
1) how to rotate infinitely?
2) how to stop?
Thank you.

kijongGil, 2018-12-03:
We will update new dynamixelSDK API(clear multi turn). You can use it our [develop-github](https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/develop)
I think it will be help you.

gooinc2, 2018-12-03:
Hello Gilbert,

Could you point out the specific link (place) for 'clear multi turn' ? Hard to search/find.

kijongGil, 2018-12-04:
Oh, sorry. you can find it. (https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/develop/c/example/protocol2.0)

kijongGil, 2018-12-06:
When you set 100 to Goal Velocity, the dynamixel has 22.9 rpm. This mean that Dynamixel rotate 22.9 turn per min.
And then I think that [this](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/264) is answer to your  another questions.

closed 2018-12-26 by gpwhd07