ROBOTIS-GIT/DynamixelSDK #263: C++ Library design issue

santdiego, 2018-11-29:
Hi. I'm getting troubles compiling the library in different Ide's. For example, if I compile the dll on Visual Studio I cannot use it in QtCreator project. Even more, If I compile the dll on QtCreator and I use it on Qt project it doesn't work correctly ( on MSVC2017 64bit or MinGW32bit is the same). The only way that works correctly is if the library and the project are compiled with the same compiler (using Visual Studio) in the same machine; other combination may not work. I think the problem relies on the header of the library ( dynamixel_sdk.h and its dependencies). It's manifested on warning C4251 which is getting in any project compilation using this library. The solution is to use d-pointers; I suggest this link to understand the problem: https://wiki.qt.io/D-Pointer

kijongGil, 2018-11-30:
Hi, @santdiego :)
Thanks for your contribution.
Actually, DynamixelSDK is optimized for visual studio. We will check it and try to solve it.

Thanks,
Gilbert

gpwhd07, 2019-01-29:
@kijongGil 
How is this update going?

santdiego, 2019-02-11:
Since this problem affects the portability of the library, and also requires recompiling the library and all the dependent units each time a source code change is made, I see no advantage in maintaining a dynamic library structure. I would strongly recommend migrating to a "header only" structure, since it facilitates the compiling and configuring process, and also generates highly optimized binaries.

sunghowoo, 2025-06-20:
This issue is quite old, so we will close it for now. If the problem still exists, please feel free to reopen it.

closed 2025-06-20 by sunghowoo