ROBOTIS-GIT/DynamixelSDK #264: Present current and velocity

lenonrobot, 2018-12-02:
Hi, I am using the motor XM430-w350-R, The python programe of DynamixelSDK is used (protocol 2, see part of codes as follows). I wonder how to calculate the current and velocity feedbacks from the read values (from ADDR_PRO_PRESENT_CURRENT   = 126, ADDR_PRO_PRESENT_VELOCITY   = 128)

DXL_ID                      = 2                
DXL_MINIMUM_POSITION_VALUE  = 1500           # Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 2500 

ADDR_PRO_PRESENT_CURRENT   = 126
ADDR_PRO_PRESENT_VELOCITY   = 128

dxl_goal_position = [DXL_MINIMUM_POSITION_VALUE, DXL_MAXIMUM_POSITION_VALUE]       

I modified read_write.py, let the motor rotate between 1500 and 2500, and tried read present current or velocity

dxl_present_cur, dxl_cur_result, dxl_cur_error = packetHandler.read2ByteTxRx(portHandler, DXL_ID, ADDR_PRO_PRESENT_CURRENT)
        if dxl_cur_result != COMM_SUCCESS:
            print("%s" % packetHandler.getTxRxResult(dxl_cur_result))
        elif dxl_cur_error != 0:
            print("%s" % packetHandler.getRxPacketError(dxl_cur_error))
print("[ID:%03d] GoalPos:%03d  PresCur:%03f" % (DXL_ID, dxl_goal_position[index], dxl_present_cur))

print result:

[ID:002] GoalPos:2500  PresCur:23.000000
[ID:002] GoalPos:2500  PresCur:11.000000
[ID:002] GoalPos:2500  PresCur:12.000000
[ID:002] GoalPos:2500  PresCur:12.000000
[ID:002] GoalPos:2500  PresCur:11.000000
[ID:002] GoalPos:2500  PresCur:11.000000
[ID:002] GoalPos:2500  PresCur:65460.000000
[ID:002] GoalPos:2500  PresCur:65509.000000
[ID:002] GoalPos:2500  PresCur:1.000000
Press any key to continue! (or press ESC to quit!)
[ID:002] GoalPos:1500  PresCur:65496.000000
[ID:002] GoalPos:1500  PresCur:65516.000000
[ID:002] GoalPos:1500  PresCur:65517.000000
[ID:002] GoalPos:1500  PresCur:65518.000000
[ID:002] GoalPos:1500  PresCur:65519.000000
[ID:002] GoalPos:1500  PresCur:65519.000000
[ID:002] GoalPos:1500  PresCur:83.000000
[ID:002] GoalPos:1500  PresCur:21.000000
[ID:002] GoalPos:1500  PresCur:65535.000000

For velocity:
        dxl_present_vel, dxl_vel_result, dxl_vel_error = packetHandler.read4ByteTxRx(portHandler, DXL_ID, ADDR_PRO_PRESENT_VELOCITY)
        if dxl_vel_result != COMM_SUCCESS:
            print("%s" % packetHandler.getTxRxResult(dxl_vel_result))
        elif dxl_vel_error != 0:
            print("%s" % packetHandler.getRxPacketError(dxl_vel_error))

print("[ID:%03d] GoalPos:%03d  PresVel:%03d" % (DXL_ID, dxl_goal_position[index], dxl_present_vel))

print result:

[ID:002] GoalPos:2500  PresVel:087
[ID:002] GoalPos:2500  PresVel:188
[ID:002] GoalPos:2500  PresVel:189
[ID:002] GoalPos:2500  PresVel:189
[ID:002] GoalPos:2500  PresVel:190
[ID:002] GoalPos:2500  PresVel:190
[ID:002] GoalPos:2500  PresVel:171
[ID:002] GoalPos:2500  PresVel:078
[ID:002] GoalPos:2500  PresVel:024
Press any key to continue! (or press ESC to quit!)
[ID:002] GoalPos:1500  PresVel:4294967189
[ID:002] GoalPos:1500  PresVel:4294967108
[ID:002] GoalPos:1500  PresVel:4294967107
[ID:002] GoalPos:1500  PresVel:4294967107
[ID:002] GoalPos:1500  PresVel:4294967107
[ID:002] GoalPos:1500  PresVel:4294967107
[ID:002] GoalPos:1500  PresVel:4294967133
[ID:002] GoalPos:1500  PresVel:4294967232
[ID:002] GoalPos:1500  PresVel:4294967277

I wonder how to calculate the current and velocity feedbacks from the read values (from ADDR_PRO_PRESENT_CURRENT   = 126, ADDR_PRO_PRESENT_VELOCITY   = 128)

Thanks.

kijongGil, 2018-12-03:
Hi @lenonrobot :)
I think present current and velocity value are correct. Just take these two values of two's complement. 

as like:
if dxl_present_velocity > 0x7fffffff:
       dxl_present_velocity = dxl_present_velocity - 4294967296
 if dxl_present_current > 0x7fff:
        dxl_present_current = dxl_present_current - 65536

Thanks,
Gilbert.

gpwhd07, 2018-12-26:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-12-26 by gpwhd07

rohitz123, 2021-05-23:
hey guys,
I have XL430 motor which is connected to U2D2 the problem I am facing is, I am trying to get load for the joint, but I am facing an error. Here the code and Language is  python.
code:
dxl_current = packetHandler.read2ByteTxRx(portHandler, ID, load_address)
print("PresCurrent: %03f" % (dxl_current ))

result:
Traceback (most recent call last):
  File "func.py", line 437, in <module>
    start.get_load(ID)
  File "func.py", line 329, in get_load
    load(ID)
  File "func.py", line 106, in load
    dxl_current_error,dxl_present_current = packetHandler.read2ByteTxRx(portHandler, ID, load_address)
ValueError: too many values to unpack (expected 2)

I cant solve this problem, please helpme out!

ROBOTIS-Will, 2021-05-24:
Hi @rohitz123 
The `read2ByteTxRx` method returns 3 values : data, result, error
Please rectify your code to receive these items correctly.
Refer to below thread as an example.
https://stackoverflow.com/questions/1479776/too-many-values-to-unpack-exception