ROBOTIS-GIT/DynamixelSDK #265: Reading CURRENT_SPEED register on AX12W while operating in wheel mode

lullabee, 2018-12-02:
Hi
I'm using `DynamixelSDK main branch and AX12W servo. I am using the C++ version of the SDK.
I set the motors on wheel mode and I am able to control the speed (using GOAL_SPEED register). However when I try to read the CURRENT_SPEED it seems that the values are sometimes correct and sometimes completely wrong (ie from 5% to 100% of the total speed)

kijongGil, 2018-12-03:
Hi @lullabee :)
I'm sorry, I don't understand your situation. Because when you set the motors on wheel mode, maybe you set the Moving Speed register. So, when you read the CURRENT_SPEED, this is Moving Speed value you set.

Could you tell me more?

Thanks,
Gilbert

lullabee, 2018-12-03:
Yes I set wheel mode then moving speed.
Current_speed should return the current speed not the one is set. 1 out of
5 readings (100hz) returns an invalid value.

For instance let say I write 300 to moving speed. Current speed will be
sometimes around 300 but sometimes would just return a random value. No
error is printed

Same on multiple ax12w. Not for a particular one.

kijongGil, 2018-12-04:
What did you use baudrate? I tested checking current speed on 57600 bps. All of current speed is fine.

lullabee, 2018-12-04:
I used 57600 just now and still have the same issue
As an example
[image]
It seems like I get wrong readings at a periodic rate. Same on my 3 motors. Period is the same

lullabee, 2018-12-04:
[image]
Same plot with MX12

lullabee, 2018-12-04:
Turns out that at very low speed all the servos also seem to slow down.

closed 2018-12-04 by lullabee