ROBOTIS-GIT/DynamixelSDK #275: [3.6.0][MATLAB][USB2Dynamixel2][MX-64] Is there anyway to read Current Torque in matlab ?

kinter107, 2019-01-01:
I'm trying to read present torque of mx-106 in matlab but i ended up with no success because there is no such option in it's e-manual table.
Then one of my friends suggested to read " present Current " and calculate torque from that But The values of present current is " integer " which causes the reduction of precision.
Can someone plz help me with this ?
thanks in advanced.

kijongGil, 2019-01-02:
Hi, @kinter107 :)
I hope that [this](http://emanual.robotis.com/docs/en/dxl/mx/mx-106/#current-68) is will help you.

Thanks, 
Gilbert

kijongGil, 2019-01-17:
I'm sorry for late reply.
You don't need to downgrade dynamixel's firmware from protocol 2 to 1.
I gave you incorrectly link. protocol 2.0 control table is [here](http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/#control-table).
Load is not torque. If you want to get torque, you have to calculate from present current.

closed 2019-01-24 by kijongGil