ROBOTIS-GIT/DynamixelSDK #279: [3.6.0][MATLAB][USB2Dynamixel2][MX-106][Protocol 1.0] Torque enable command in matlab using protocol 1.0 doesn't work !

kinter107, 2019-01-05:
Hi,
I use the code below (.txt file)  which is made based on "read-write.m" example in sdk, the only difference is that I send velocity command instead of position command and when I run my code Torque enabling doesn't work at all !

[test.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2729450/test.txt)

I use r+ manager 2.0 and everything works fine and when I enable Torque servo locks moving by hand , but in code no such thing happens.
any help is greatly appreciated.

kijongGil, 2019-01-10:
Hi @kinter107 :)
I'm sorry for late reply.
Did you run our example code as read_write.m?

kinter107, 2019-01-12:
Hi @kijongGil,
yes i ran the example but it didn't work.
after couple of tries , i figured out that even in protocol 1 , i should use address 64 for torque enabling which is in protcol 2 control table !!! That's wierd.
i checked it multiple times with R+manager to ensure that each dynamixel is using protocol 1.
Plz Help ! i need it for the paper i'm about to submit :((((1

kijongGil, 2019-01-17:
I need an error log and more detail your situation.
What is your OS, how to build DynamixelSDK, what is version your matlab etc...

closed 2019-01-24 by kijongGil