ROBOTIS-GIT/DynamixelSDK #281: [3.6.0][python][USB2Dynamixel2][XM430-W350-R] packetHandler.read2ByteTxRx returns huge result when it should be negetive

d0ngha0, 2019-01-17:
I have some problems when I using Dynamixel_SDK python version, it will be very helpful if you can give me a hand.
I have tried to use 'packetHandler.read2ByteTxRx' to get the present current, it seems that when the current is positive quantity, I get the good result, but when the current is negetive quantity, the result is always shows like 65527, 65529, which is apparently wrong. The same situation shows when I use 'packetHandler.read4ByteTxRx' to get the present velocity and position.
Is there any codes to fix this? Or if it is possible to show me some examples to read present velocity, position, current.

kijongGil, 2019-01-17:
Hi @d0ngha0 :)
I think [this](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/264) will be help you.

Thanks,
Gilbert

d0ngha0, 2019-01-17:
Thank you, kijongGil!
It solves problems perfectly!

closed 2019-01-17 by d0ngha0