ROBOTIS-GIT/DynamixelSDK #284: [3.7.0][MATLAB][USB2Dynamixel2][MX-64AT] Sinusoidal Time varying profile generation using MATLAB

shulnak09, 2019-02-01:
Hi,

My project requires to control dynamixel using time varying angular velocity , instead of constant RPM. Suppose, I have to oscillate sinusoidaly between max and min goal position continuously for a period of five seconds. How to execute it in MATLAB?

kijongGil, 2019-02-11:
Hi @shulnak09 :)
I'm sorry for late reply.
We have examples for control DXL with MATLAB.
Refer to [this](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#matlab-windows).
I recommend that run the read-write example, then make a code by yourself.

Thanks,
Gilbert

shulnak09, 2019-02-12:
Thank you Gilbert for the reply.

I have followed the Matlab example and have successfully run the readwrite.m (Protocol1.0) example controlling both the speed of motor and the servo motion between min and max position.

Here is a link to the code and on lines 135 and 136 there is the code for goal position and speed control taking hint from previous issues.

[https://drive.google.com/file/d/1dqIkyG51fr0nFjR940M5zUfX7wLR84r7/view?usp=sharing](url)

But my objective is not to reach the goal position at constant speed but rather with variable speed that changes as a sine function. Is it possible to modify the code or write a loop to have an array of input command between min and max goal position with different speed?

Regards
Anshul

kijongGil, 2019-02-15:
I understand what you want. You can control to velocity with Goal velocity.
Refer to [this](http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#goal-velocity).
Maybe you have to change goal velocity  at every time as like write2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, 104, velocity);

closed 2019-02-27 by gpwhd07