ROBOTIS-GIT/DynamixelSDK #286: Reading AX-12A Status Packet (Python, Raspbian, Raspberry Pi 3B)

CalvinBarajas, 2019-02-19:
Hello, I purchased 4 AX-12A actuators and I'm trying to capture a status packet using Raspberry Pi 3B. I'm able to successfully communicate with the AX-12A. I can move to goal positions successfully and control many parameters from the control table. The code below works really well for sending instruction packets. I just need sample Python code for reading the status packets. I'm using an octal tri-state buffer on a breadboard (74LS241) for half-duplex communication.

============== source code ======================
import serial
import time
import RPi.GPIO as GPIO

channel = 18

GPIO.setmode(GPIO.BCM)
GPIO.setup(channel, GPIO.OUT)

ser = serial.Serial("/dev/ttyS0", baudrate=1000000, timeout=3.0)

GPIO.output(channel, GPIO.HIGH)
ser.write(bytearray.fromhex("FF FF 01 05 03 1E 32 03 A3"))
time.sleep(0.1)
GPIO.output(channel, GPIO.LOW)
time.sleep(3)

GPIO.output(channel, GPIO.HIGH)
ser.write(bytearray.fromhex("FF FF 01 05 03 1E CD 00 0b"))
time.sleep(0.1)
GPIO.output(channel, GPIO.LOW)
time.sleep(3)
    
ser.close()
GPIO.cleanup()

ROBOTIS-zerom, 2019-04-03:
Hello, Calvin.
Examples of read/write using the Dynamixel SDK written in python, see the link below:
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/python_read_write_protocol_1_0/#python-protocol-10

ROBOTIS-zerom, 2019-05-31:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-05-31 by ROBOTIS-zerom

pratapbhanusolanki, 2019-06-14:
Hey @CalvinBarajas, were you able to solve your problem. 
I am on a similar boat. I am trying to use the DynamixelSDK library, however, it does not take care of the (74LS241) for half-duplex communication.

CalvinBarajas, 2019-06-15:
Hi Pratap, check out my video series on YT. It covers a lot of material:
https://www.youtube.com/watch?v=O8mVpsrBRhg&t=40s
It covers, how to setup 74LS241, how to wire everything up, etc. In order to get the status packet, you need to capture the stream after the LOW statement.
Sincerely,Calvin

pratapbhanusolanki, 2019-06-17:
Hey Calvin, I decided to go with the U2D2 converter with my Beaglebone as it is working good with the Dynamixel library. I cannot use your code directly as it is catered for the Raspberry Pi. 

However, Thank you for the tutorial. I like the detailed description of the components, it is helpful for my project. 

Best regards, 
Pratap

CalvinBarajas, 2019-06-17:
Thanks Pratap. I'm glad you got it working and I'm glad I could help.
Sincerely,Calvin

Beaglebone