ROBOTIS-GIT/DynamixelSDK #288: [3.7.0][PYTHON][USB2Dynamixel2][XM430-W350-R]reduce USB latency time

bishopAL, 2019-02-25:
Hi! I have some problems on USB latency time. I have modified latency_time to 1ms in Ubuntu. What I need to do is to get the present position and present velocity, then set the torque, all of these operations costs 6ms (with PacketHandler.readTxRx and PacketHandler.writeTxRx function). It's fast enough for 1 motor, but I have 12 motors to drive. It will cost 72ms for a period. That's disaster! So how can I reduce this latency time? Will ROS be helpful?

rdbrewer, 2019-03-05:
I think you have to set the latency timer both in the Dynamixel SDK and in the OS. For the SDK, edit the file: \DynamixelSDK-3.7.0\python\src\dynamixel_sdk\port_handler.py

There's a line "LATENCY_TIMER = 16", which I changed to 1 and then re-installed the module.

bishopAL, 2019-03-05:
Thanks! I have tried it and it's working.

closed 2019-03-05 by bishopAL