ROBOTIS-GIT/DynamixelSDK #290: [3.7.0][Python][U2D2][XM540-W270-R] Reading PresentPosition (PortRegister = 132) corrupts data when reading other variables

rdbrewer, 2019-03-05:
Motor: XM540-W270-R
Language/SDK: DynamixelSDK-3.7.0 in Python version 2.7
OS: 64-bit Windows 8.1
Connecting to the motor over a U2D2 serial converter at 4MBPs baud.
StatusReturnLevel = 1. 

When I read just the PresentPosition (PortRegister = 132, ReadLength = 4), the values are correct. When I read just the voltage (PortRegister = 144, ReadLength = 2) and temperature (PortRegister = 146, ReadLength = 1), they are correct. However, when I read all 3 variables, all values are incorrect.  Also, reading the temperature should require only 1 byte, but it requires 2. 

I'm attaching a simply python file which reads the PresentPosition, PresentInputVoltage, and PresentTemperature in a loop. 

[test_read_debug.pdf](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2928890/test_read_debug.pdf)
[test_read_debug.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/2928891/test_read_debug.txt)

ROBOTIS-Will, 2019-04-19:
@rdbrewer 
I'm so sorry for the delayed response.
Our engineers are investigating on this issue and they'll get back to you very soon.
Thank you very much for your patience

kijongGil, 2019-04-19:
@rdbrewer 
I tested test_code_debug you uploaded on both windows 10 and ubuntu 18.04.
It works well without issue.

Could you tell me more information about your environment? Then, how about change cable?

Thanks,
Gilbert.

ROBOTIS-zerom, 2019-04-24:
This issue was identified as a problem with the usb latency timer.

closed 2019-04-24 by ROBOTIS-zerom