ROBOTIS-GIT/DynamixelSDK #296: Speed control for Dynamixel X series

sonary01, 2019-04-01:
Hi, 
I'm using XL-430-W250 motors and programming in Matlab. I want to run at a lower speed. Previously I had used the MX series and able to run at different speeds. What is the equivalent of 'Moving Speed (32)' for X series? I tried to change the Goal Velocity (104). But, it runs at the speed! How to change the speed of XL430 motor?
`write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, 104, typecast(int32(10), uint32'));`

t-kitajima, 2019-04-02:
Hi,
To change the speed of the X series, set the value for Goal Velocity (104).
It seems to set the target speed to Goal Velocity (104).
There is no single quote attached, does not matter?
```
write4ByteTxRx( port_num, PROTOCOL_VERSION, DXL_ID, 104, typecast( int32( 10 ),  uint32' ) );
```

sonary01, 2019-04-02:
That's a typo. Corrected in the question. There is no change in speed yet. The 'present velocity' reading while the motor runs between 100 to 4000 varies from 40 to 298. Anything else can be done to reduce the speed?

ROBOTIS-zerom, 2019-04-03:
In position control mode & extended position control mode, Profile Velocity(112) can be used the maximum velocity of profile.

profile maximum velocity = (profile velocity(112) value) * 0.229 rpm

Try setting the profile velocity value to a value as low as 20.

sonary01, 2019-04-03:
Thanks!! It worked.

closed 2019-04-03 by sonary01

AnastasiyaRybakova, 2021-06-07:
Hello, dear Sonia,
I hope you are doing well.
I would like to ask one moment, if you don't mind. Did you try to make the velocity controlling for the manipulator robot or for the one which uses motors for the navigation task?
I am working with the manipulators and I am trying to find out the way of the position and velocity controlling at the same moment, but I am a bit confused is it possible or not? 
Some sources I have seen mentioned that the velocity controlling with the position controlling is not kind of possible. 

If you have a chance to share your mind and experience, I would like to know it. 
I will truly appreciate it.
Sincerely,
Anastasiya

ROBOTIS-Will, 2021-06-07:
Hi @AnastasiyaRybakova 

I'll assume that you have DYNAMIXEL X series (for example, [XM430-W350](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350)).
When DYNAMIXEL is set to [Position Control Mode](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11), you can limit the maximum speed(velocity) and acceleration of the profile using [Profile Velocity(112)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#profile-velocity112) and [Profile Acceleration(108)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#profile-acceleration108)
Based on the profile acceleration and profile velocity values, different type of profile will be created automatically.
[image]

Please refer to below eManual for more details.
https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#what-is-the-profile

mink007, 2022-05-05:
In the documentation it says '0' represents infinite acceleration or velocity.
 What does '1' represent? and what does the maximum value (32767) 'represent' ?

Does '1' represent minimum acceleration/velocity and 32767 the maximum?

ROBOTIS-Will, 2022-05-09:
Setting the Profile Acceleration / Velocity `0` with the Velocity-based Profile, it represents the maximum acceleration / velocity that the motor can physically run.
Setting other values than 0 can be calculated with the given unit values in the eManual.
For example, setting the Profile Acceleration to `1` will accelerate or decelerate the core motor at 214.577 rev/min^2.
Although the range of these values can be set up to 32,767, at some point, the motor won't be able to catch up the set value.

mink007, 2022-05-09:
I essentially want the "XM430-W350-T" motor speed to be 1 revolution per minute to have a smooth and appealing to eye (natural) motion of the arm.

What number I should put in XXX in below command that will help in obtaining 1 rev/min ? 
write4ByteTxRx( port_num, PROTOCOL_VERSION, DXL_ID, 104,  XXX );

I am using SDK readwrite (C++) example and modifying it to my needs.

shiba-8ro, 2022-05-11:
@mink007 
Goal Velocity (104) is the command value for Velocity control mode.
So you need to use Profile [Velocity (112)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#profile-velocity112).
Profile Velocity (112) and Profile Acceleration(108) must be sent to Dynamixel before  Goal Position (116) sending.
Set Profile Velocity (112) and Profile Acceleration(108) ==> Set Goal Position (116) ==> Profile generation and Moving start

mink007, 2022-05-12:
@shiba-8ro will try 112 & 108 and then 116. Thank you.