ROBOTIS-GIT/DynamixelSDK #301: Setting up DynamixelSDK (c++ language) for Raspberry Pi

ShivangiMisra12, 2019-04-23:
Preliminaries:
- I'm using `USB2Dynamixel2` serial converter
- I'm using `XL430-W250-T`

I'm trying to set up Dynamixel SDK on a Raspberry Pi 3B+ running Raspbian Stretch OS. I'm running into an issue when setting up the library for c++ code (gcc version 6.3.0) at the step:

`$ sudo apt-get install gcc-multilib g++-multilib` 

It returns:
`E: Package 'gcc-multilib' has no installation candidate`
`E: Unable to locate package g++-multilib`
`E: Couldn't find any package by regex 'g++-multilib'`

Any suggestions to complete the library setup and overcome this problem would be appreciated.

GeissonR, 2019-05-09:
Same problems here!

GeissonR, 2019-05-10:
> Preliminaries:
>
> * I'm using `USB2Dynamixel2` serial converter
> * I'm using `XL430-W250-T`
>
> I'm trying to set up Dynamixel SDK on a Raspberry Pi 3B+ running Raspbian Stretch OS. I'm running into an [...]

I skipped this step and it worked fine on my Raspberry Pi 3

ROBOTIS-zerom, 2019-06-04:
gcc-multilib & g++-multilib does not exist in raspbian.
skip that.

closed 2019-06-04 by ShivangiMisra12

robotiscustomer, 2020-10-23:
Hi, I got the same error when installing the SDK on Ubuntu 20.10 Desktop on my Pi. After I found this thread, I ignored the errors. I then cd to /c/build/linux_sbc  and make.  I got 
"make: *** [Makefile:68: livdxl_sbc_c.so] Error 1". Next, I typed "make clean && make" but the same error remains. How do I fix it?