ROBOTIS-GIT/DynamixelSDK #302: Reading operation in dynamixel motors is too slow.

santdiego, 2019-04-23:
Hi. 
Im working with :
-Windows 10 x64- Microsoft Visual C++ Compiler 15.0 
-Robotis H-Manipulator. 
Setup:
-Motors baudrate are set to 4Mbps. 
-USB latency timer set to 1msec; 
-Windows Timer Resolution set to 1msec ( with timeGetDevCaps)

Results: 
Time for syncRead operation (6 motors)>4ms
Time for individual read operation (6 motors)>6ms
The measurements are consistent whether the DynamixelSDK is used or not (sending the frame with the blocking versions of the WriteFile / ReadFile functions).

At 4 Mbps I would expect much faster communication .Is there any communication delay added in the firmware of the motors?

ROBOTIS-zerom, 2019-05-31:
Hi, Santdiego.
The item in the control table that affects communication delay is "Return Delay Time". 
The default value of "Return Delay Time" is 500 [μs].
Return Delay Time : 6 motors x 0.5ms = 3ms

So, change the "Return Delay Time" of all motors to 0 and retest.

santdiego, 2019-05-31:
Hi zerom. Of course I did that procedure, however the loss of packages produced that the delay was in those values. After spending a long time with the problem I realized that it is a driver failure. When I decrease the latency of the usb from the device manager from 16 to 1 the described error occurs, however when I do it gradualy there is no error. It is quite rare and I do not know the cause. I could not achieve asynchronous reading either. I will continue investigating the problem.
Currently I am achieving communication around 1ms minimum, without loss of packages.
I would like to emphasize that on my PC I have installed several devices that use the FTDI driver, in particular there was a driver conflict with the "Novint Falcon" when installing the R + Launcher, or the Novint stopped working or the dynamixel stopped working . The FTDI driver is always conflicting.

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will