ROBOTIS-GIT/DynamixelSDK #303: [3.7.0][MATLAB][USB2Dynamixel2][XL430-W2250] Issue : groupSyncWrite not in library

bambz96, 2019-04-25:
Hi there, 
I'm trying to use the Group Sync Write functionality however I'm getting an error in MATLAB saying the method was not found (see below). 
[image]
I did a library function search to see whether the method is listed but noticed it isn't and there aren't any of the group functions available (see below) 
[image]
I've also tried using the example Matlab script but I'm getting the same issue. Also note that other functionality is fine, I have no trouble sending individual commands however I'd like to use the Sync functionality to speed things up. 
Thanks, 
Bambz

ROBOTIS-zerom, 2019-06-04:
Hi, Bambz.

I have found that the library file & dll file in C language is outdated. 
I rebuild and uploaded it to the latest file.
The old file is missing the "clearMultiTurn" function.

Try again with the latest in the master branch.

ROBOTIS-zerom, 2019-06-14:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-06-14 by ROBOTIS-zerom

KartikeyaWalia, 2022-08-02:
Hello,

I am having the same issue and with the latest version of the SDK it is still showing 'group SyncWrite' isn't there in the library. I have checked and verified that it is missing. 
[image]
Please respond.