ROBOTIS-GIT/DynamixelSDK #304: [Arduino][OpenCR][MX-28 & MX-106] Issue: No able to run my code with user-functions

aaceve, 2019-05-01:
Hi,

I'm using the latest Arduino IDE with the OpenCR board. Everything runs ok if I write my program in a single piece of code inside the setup() or loop() function. But when I try to write a user function, the communications amoung board and servos gets stuck.

How can I properlly write user-functions using the DinamixelSDK.h (not the DynamixelWorkbench.h)? 

As far as I could understand from examples, the main program has 5 simple steps:
1.- Definitions of variables
2.- Open the port
3.- Set port baudrate
4.- Dynamixel command: like Ping, etc
5.- Close the port

Instead of doing these 5 steps everytime inside a user-callable function, I wanted to declare variables as globals and then call them inside my user-function, but it didn't work. The program compiles, but gets stuck when I run it.

Here, I'm attaching two version of the same program and its outputs. The program named ping.ino just call the ping method 18 times inside a for loop and it works perfect. In this one the code was written following the same structure of examples. The program named "modified_ping.ino" has the same objective but by using global variables. It compiles without issues, but it doesn't work, something gets stucks.

Please, Could somebody tell me what I'm doing wrong? What I should to do in order to get my user function work properly? Thanks in advance.

[ping.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/3133492/ping.txt)
[image]
[image]
[modified_ping.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/3133493/modified_ping.txt)

OpusK, 2019-06-17:
Hi, @aaceve 

The problem is that `packetHandler` was a null pointer.
Anyway, this is not recommended for the Arduino compilation method, nor for the general programming method, for global variable initialization as function execution.

See below for Arduino related information.
 - [Pre-Processing](https://github.com/arduino/Arduino/wiki/Build-Process#pre-processing)

We recommend that you change the code as shown below.
In addition, when using pointers, be sure to include the code to check for nullptr.

```cpp
#include <DynamixelSDK.h>

// 1.- Definitions as globals
dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler;

...

void setup() {
  portHandler = dynamixel::PortHandler::getPortHandler("OpenCR_DXL_Port");
  packetHandler = dynamixel::PacketHandler::getPacketHandler(1.0);

...

```

aaceve, 2019-06-21:
Hi Kei,
Yes, it works just perfect. Thank you.

closed 2019-07-01 by ROBOTIS-zerom