ROBOTIS-GIT/DynamixelSDK #306: [3.5.2][Arduino][OpenCM904][XL430-W250-T] Not being able to change ID.

Crodu, 2019-06-01:
As i am not too experienced on the topic, i've been trying to figure out the DYNAMIXEL XL430-W250-T for a while.

Currently i can write data using a slightly modified version of the protocol 2.0 read_write_x program.

According to the control table:

Address: 7 
Size (bytes):  1 
Name:  ID 
Description:  Dynamixel ID 
Access: **RW** 
Default value:  1

But as i run the command:

`dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, 1, 7, 3, &dxl_error);`

It does not change my dynamixel at current id **1**, to **3** through address **7** and throws the following error:
`[RxPacketError] Writing or Reading is not available to target address!`

While the address is clearly "RW".

ROBOTIS-OFFICIAL, 2019-06-03:
Hi, Crodu.
Address 7 (ID) is the EEPROM area. 
Data in the EEPROM area can only be written to if Torque Enable(64) is cleared to '0' (OFF).

Make sure that Torque Enable(64) is cleared to '0'.
If the same problem occurs when Torque Enable(64) is cleared to '0', please let me know again.

ROBOTIS-zerom, 2019-06-14:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-06-14 by ROBOTIS-zerom

dvlprjrlvno, 2021-01-19:
Hi, i've been trying to read the present position or the ping instruction of the DYNAMIXEL XL430-W250-T, but I have not been able to read the current position of the servo. The servo does not respond to the read instruction.

There is some register that needs to be configured to activate the read operation. 

Thank you.

dvlprjrlvno, 2021-01-19:
I am executing this instruction in arduino due to read the present position:

0xFF | 0xFF | 0xFD | 0x00 | 0x01 | 0x07 | 0x00 | 0x02 | 0x84 | 0x00 | 0x04 | 0x00 | 0x1D | 0x15

ROBOTIS-Will, 2021-01-20:
Hi, @dvlprjrlvno 
XL430-W250-T uses 5V level TTL communication, which Arduino DUE(3.3V) does not support.
Please note that we provide Arduino library for DYNAMIXEL control so that customers won't have to go through hard time communicating with DYNAMIXEL.
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino
If you are using DYNAMIXEL Shield, you can use DYNAMIXELShield Arduino library
https://github.com/ROBOTIS-GIT/DynamixelShield
Thank you.

dvlprjrlvno, 2021-01-20:
Hi, ROBOTIS-Will
Yes, I use a TTL 3.3 to 5 vdc level converter. I'm also using an XL-320, and it responds correctly to read and ping instructions. But the strange thing is that the XL-430 does not respond to the read and ping instruction. With the write instruction I have no problem, the XL-430 responds correctly.
I'm going to follow your recommendation to use the Dynamixel Shield, although it would be interesting to get the XL-430 working. This to verify the information in the user manual.
I was expecting my error to be the configuration of an EEPROM address or register.
Thank you.

ROBOTIS-Will, 2021-01-21:
Some embedded board may have a slight error in buadrate. Usually [less than 3% of error is acceptable](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#baud-rate8), but changing buadrate something like 57600bps may resolve the communication issue.
Please note that the default baudrate of XL-320 is 1,000,000 bps while other X series is 57,600 bps.
Thank you.